esp8266/Arduino #5725: Webserver hangs up while device is still reachable by ping without problems

calkoe, 2019-02-06:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			        [esp8266EX]
- Core Version:			(#5724)
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB]
- lwip Variant:                 [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:            [115200]

### Problem Description

- After a while (20 - 60sec) the Webserver stops answering to requests while the device is still     reachable by ping.
- After about 30sec the Webserver continues working without any problems.
- there are not WiFi-Scans or STA the ESP tryes to connect, see simple script below.

[image]

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
ESP8266WebServer server(80);

void handleRPC() {
  server.send(200, "application/json", "{\"id\": 2858,\"method\": \"system.getStatus\",\"payload\": {\"system\": {\"batteryPowerSupply\": true,\"batteryPercent\": 100,\"totalBytes\": 2949250,\"usedBytes\": 1004,\"freeHeap\": 35552,\"currentTaskTime\": 129867,\"status\": 5},\"wifi\": {\"ssid\": \"Wolke\",\"status\": 3,\"mode\": 3,\"ip\": \"192.168.135.163\",\"rssi\": -47},\"application\": {\"name\": \"radar\",\"version\": \"0.1\",\"statusEnable\": false,\"statusStatus\": false,\"statusText\": \"\"}},\"success\": true,\"status\": \"ok\"}");
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");
  WiFi.softAPConfig(IPAddress(192, 168, 5, 1), IPAddress(192, 168, 5, 1), IPAddress(255, 255, 255, 0));
  WiFi.softAP("arduinoOS");

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.on("/rpc", handleRPC);
  server.begin();
  Serial.println("HTTP server started");
}

void loop() {
  server.handleClient();yield();
}
```

devyte, 2019-02-08:
Please test with the following before calling softAPConfig():
```cpp
WiFi.persistent(false);
WiFi.mode(WiFI_AP);
```

and also make sure to do a full flash wipe before flashing this sketch.

devyte, 2019-10-31:
No feedback in several months. Closing.

closed 2019-10-31 by devyte