esp8266/Arduino #5727: Compiling error with V2.5

alexis69FR, 2019-02-06:
error compiling  with 2.5.0 version, run well with 2.5 beta3

NodeMcu V3
[image]

[image]

d-a-v, 2019-02-07:
> note: no known conversion for argument 1 from 'IPAddress' to 'const char*'

Depending on what you need,
you can use `ip.toString().c_str()` which is portable,
or use legacy `uint8_t operator []` => `ip[n]` which is also portable and means something else,
An `operator const uint8_t*()` may be added (but probably not `const char*`)

More details would be needed for better support, maybe from a Blynk developper, maybe for a longer cut'n'paste of your error messages.

danobot, 2019-02-07:
Hi,
I am facing the same compile error. Was reported [on Eclipse forums](https://www.eclipse.org/forums/index.php/t/1096821/) as well.

```
19:08:46 **** Incremental Build of configuration esp_release for project esp_eink_dashboard_sloeber ****
"C:\\tools\\sloeber4.3.1\\arduinoPlugin\\tools\\make\\make" -j all 
'Building file: ..\Main.cpp'
'Building file: ..\test\Component_test.cpp'
'Starting C++ compile'
"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/include" "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/lwip2/include" "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include" "-IG:/Repos/sloeber-workspace-new/esp_eink_dashboard_sloeber/esp_release/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=1 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DARDUINO=10802 -DARDUINO_ESP8266_WEMOS_D1R1 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="\"ESP8266_WEMOS_D1R1\""   -DESP8266   -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266" -I"G:\Repos\sloeber-workspace-new\esp_eink_dashboard_sloeber\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\variants\d1" -I"G:\Arduino_Hardware\Jantje_Arduino_libraries\gdb" -I"G:\Arduino_Hardware\Jantje_Arduino_libraries\gtest" -I"C:\tools\sloeber4.3.1\arduinoPlugin\libraries\Adafruit_GFX_Library\1.3.6" -I"C:\tools\sloeber4.3.1\arduinoPlugin\libraries\ArduinoJson\5.13.4\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\libraries\GxEPD2\1.1.1\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WebServer\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SPI" -MMD -MP -MF"Main.cpp.d" -MT"Main.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\Main.cpp"  -o  "Main.cpp.o"
'Starting C++ compile'
"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/include" "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/lwip2/include" "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include" "-IG:/Repos/sloeber-workspace-new/esp_eink_dashboard_sloeber/esp_release/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=1 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DARDUINO=10802 -DARDUINO_ESP8266_WEMOS_D1R1 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="\"ESP8266_WEMOS_D1R1\""   -DESP8266   -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266" -I"G:\Repos\sloeber-workspace-new\esp_eink_dashboard_sloeber\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\variants\d1" -I"G:\Arduino_Hardware\Jantje_Arduino_libraries\gdb" -I"G:\Arduino_Hardware\Jantje_Arduino_libraries\gtest" -I"C:\tools\sloeber4.3.1\arduinoPlugin\libraries\Adafruit_GFX_Library\1.3.6" -I"C:\tools\sloeber4.3.1\arduinoPlugin\libraries\ArduinoJson\5.13.4\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\libraries\GxEPD2\1.1.1\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WebServer\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src" -I"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\SPI" -MMD -MP -MF"test\Component_test.cpp.d" -MT"test\Component_test.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\test\Component_test.cpp"  -o  "test\Component_test.cpp.o"
'Building file: ..\test\Composite_test.cpp'
'Building file: ..\test\Leaf_test.cpp'
'Starting C++ compile'
"C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/include" "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/lwip2/include" "-IC:\tools\sloeber4.3.1\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include" "-IG:/Repos/sloeber-workspace-new/esp_eink_dashboard_sloeber/esp_release/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=1 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DARDUINO=10802 -DARDUINO_ESP8266_WEMOS_D1R1 -DARDUINO [... cut ...]
libraries\ESP8266WiFi\src\subdir.mk:126: recipe for target 'libraries\ESP8266WiFi\src\WiFiClient.cpp.o' failed
make: *** [libraries\ESP8266WiFi\src\WiFiClient.cpp.o] Error 1
'Finished building: C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\ESP8266WiFi\src\WiFiClientSecureAxTLS.cpp'
' '
"C:/tools/sloeber4.3.1/arduinoPlugin/tools/make/make -j all" terminated with exit code 2. Build might be incomplete.

19:08:55 Build Failed. 11 errors, 0 warnings. (took 8s.879ms)

```

Let me know if I can help more.

d-a-v, 2019-02-07:
Please try with latest release, you are using 2.5.0-beta2.
```
C:\tools\sloeber4.3.1\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries.....
```

alexis69FR, 2019-02-10:
2.5.0-beta2 and beta3 are ok for me, but not 2.5.0 release

pfeerick, 2019-02-19:
Can we get an actual [MCVE](https://stackoverflow.com/help/mcve) that triggers these errors? I'm suspecting different issues are being spoken about here... as the first is somewhat related to the Blynk library against 2.5.0-beta3... for which both the ESP8266_Standalone and ESP8266_Standalone_Manual_IP are examples are compiling just fine on the release 2.5.0. 

Without some code that triggers the specific error being faced, it's the blind leading the blind... which won't get us far! :open_mouth:

danobot, 2019-02-20:
The 2.5.0 release solved my problems

earlephilhower, 2019-02-26:
If we get no MCVE in the next few days, I think this is safe to close.  Lots of positive 2.5.0-rel responses.

earlephilhower, 2019-05-01:
No response in 2 months and some reports of 2.5.0 release solving the issue, so I'm closing.  If 2.5.0 or GIT head has an issue, please open a new report with MCVE.

closed 2019-05-01 by earlephilhower