esp8266/Arduino #5730: Upgrade to 2.5.0 problems, ip_addr incompatibility and wrong pin definitions in arduino_pins.h

codebeat-nl, 2019-02-06:
When upgrading to 2.5.0, the following problems occur when compiling:
- type ip_addr has been changed, now need to change it to ip4_addr and is no longer compatible with earlier versions.
- Pin D0 to D5 are defined as variables in arduino_pins.h however incorrect. Get macro expansion errors when compiling. After removing my defines and using these pins definitions, esp starts to reboot all the time. D0 is 0, D1 = 1, etc. This is not correct, it refers to the wrong GPIO.

This is the correct mapping I have used all of the time and works just fine:
 ```
// ESP8266 Pin names to GPIO,  weird pin mapping correction.
 // https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h#L37-L59
 // Pin numbers written on the board itself do not correspond to ESP8266 GPIO pin numbers. 
 // The defines below make using this board easier (If you want to use esp pin 5, 
 // use D5 for pin number, and it will be translated to 'real' GPIO pin 14):

#define D0 16 // supports PWM
#define D1  5 // I2C Bus SCL (clock)
#define D2  4 // I2C Bus SDA (data)
#define D3  0
#define D4  2 // Same as "LED_BUILTIN", but inverted logic
#define D5 14 // SPI Bus SCK (clock)
#define D6 12 // SPI Bus MISO 
#define D7 13 // SPI Bus MOSI
#define D8 15 // SPI Bus SS (CS)
#define D9  3 // RX0 (Serial console)
#define D10 1 // TX0 (Serial console)

#define SD0   7
#define SD1   8
#define SD2   9
#define SD3   10
#define SDCMD 11
```

d-a-v, 2019-02-06:
You don't give much details.
What was the previous core version you were using ?

>  type ip_addr has been changed, now need to change it to ip4_addr and is no longer compatible with earlier versions.

When possible, `IPAddress` variables should be used in place of any `ip_addr/_t` types where `ip.addr` or `ip->addr` were used. This is much more compatible.

Pins are defined in variant/. Did you select the wrong one ?
In arduino IDE, board selection defines pin definition.

codebeat-nl, 2019-02-06:
Previous version was: 2.4.0

Error when defining for example D0 in my h-file:
Users\Codebeat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\variants\ESPDuino/pins_arduino.h:35:22: note: in expansion of macro 'D0'

 static const uint8_t D0   = 0;

codebeat-nl, 2019-02-06:
I use ip_addr with use of stat_info:

```
String suGetClientMacAddress( SUWebServerRequest* request )
{ 
  if( !!(request) && suIsAccessPointMode() )
   {  
     struct ip4_addr*      IPaddress;
     IPAddress             ip;
     struct station_info*  stat_info = wifi_softap_get_station_info();
     String                sClientIp = suGetClientIp( request );
     
     Serial.print( "Client ip: " );
     Serial.println( sClientIp );
     
     if( sClientIp.length() > 0 )
     {
      while( stat_info ) 
      {
        IPaddress = &stat_info->ip;
        ip = IPaddress->addr;
        
        if( suIpAddressToString( ip ) == sClientIp )
        {
          return String( stat_info->bssid[0], HEX )+":"+
                 String( stat_info->bssid[1], HEX )+":"+
                 String( stat_info->bssid[2], HEX )+":"+
                 String( stat_info->bssid[3], HEX )+":"+
                 String( stat_info->bssid[4], HEX )+":"+
                 String( stat_info->bssid[5], HEX );
        } 
       
        stat_info = STAILQ_NEXT(stat_info, next);
      }
     } 
   }
  
  return "";  
}

```

d-a-v, 2019-02-06:
```
String suGetClientMacAddress( SUWebServerRequest* request )
{ 
  if( !!(request) && suIsAccessPointMode() )
   {  
     struct station_info*  stat_info = wifi_softap_get_station_info();
     String                sClientIp = suGetClientIp( request );
     
     Serial.print( "Client ip: " );
     Serial.println( sClientIp );
     
     if( sClientIp.length() > 0 )
     {
      while( stat_info ) 
      {
        IPAddress ip = stat_info->ip; // should work
        // or IPAddress ip = stat_info->ip->addr; // should work too, less portable
        // String ipString = ip.toString(); // may be useful
        
        if( suIpAddressToString( ip ) == sClientIp )
        {
          return String( stat_info->bssid[0], HEX )+":"+
                 String( stat_info->bssid[1], HEX )+":"+
                 String( stat_info->bssid[2], HEX )+":"+
                 String( stat_info->bssid[3], HEX )+":"+
                 String( stat_info->bssid[4], HEX )+":"+
                 String( stat_info->bssid[5], HEX );
        } 
       
        stat_info = STAILQ_NEXT(stat_info, next);
      }
     } 
   }
  
  return "";  
}
```

d-a-v, 2019-02-07:
> `...Arduino15\packages\esp8266\hardware\esp8266\2.5.0\variants\ESPDuino/pins_arduino.h:35:22: note: in expansion of macro 'D0'``

This missing definition that you need was fixed for everybody in #4828.
You may check `ARDUINO_ESP8266_RELEASE_2_5_0` and `coreVersionNumeric()` if you need conditional compilation (first is defined in `cores/esp8266/core_version.h`, second in `cores/esp8266/core_esp8266_version.h`, both already included in your sketch)

devyte, 2019-02-15:
Closing due to user error.

closed 2019-02-15 by devyte