esp8266/Arduino #5734: Crash when Debug port is Disabled

joysfera, 2019-02-06:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[release 2.5.0]
- Development Env:		[Arduino IDE 1.8.8]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[tried both OTA and SERIAL]
- Upload Speed:   [115200]

### Problem Description

Detailed problem description goes here.
In Arduino IDE when `Debug port` is set to `Disabled` my application crashes after fetching data via https and displaying them on an LCD matrix. If I set the `Debug port` to `Serial` in Arduino IDE then everything works properly. My application does not use any debugging stuff. It does not even use the Serial stuff, does not call `Serial.begin()`, nothing of that.

I cannot debug this because when I enable debugging and thus set the Debug port to Serial the issue disappears :-)

I know this bug report is incomplete without the source code etc. but it's kinda late at night at I am glad I managed to at least reproduce 100% that mere `Debug port` setting changes application crashing to not crashing and vice-versa.

devyte, 2019-02-08:
@joysfera [This ](https://github.com/esp8266/Arduino/blob/c08efb52fbc34b5e644d29e7ad904c0e7b2ff1ef/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h#L146)comment is relevant:
```cpp
/*
 * Since both begin() functions take a reference to client as a parameter, you need to 
 * ensure the client object lives the entire time of the HTTPClient
 */
```

HttpClient keeps a pointer to client, and tries to access it in the destructor. In your order of instantiation, client gets destroyed first. Then, when HttpClient gets destroyed, it tries to access client, but it no longer exists, so garbage => crash.

Closing due to user error.

joysfera, 2019-02-08:
@devyte thanks. I've created a PR #5739 to save others the lost time due to wrong example code.

yoursunny, 2019-03-23:
This seems to be **bad API design** rather than simply blaming on user error.
To avoid such mistakes, `HTTPClient` should take `WifiClient&` reference either in its constructor, or as a `unique_ptr`.

```
std::unique_ptr<BearSSL::WiFiClientSecure> client(new BearSSL::WiFiClientSecure);
HTTPClient http(*client);
http.begin(uri);
// unless user intentionally destructs ‘client’, it’s safe
```

```
HTTPClient http;
std::unique_ptr<BearSSL::WiFiClientSecure> client(new BearSSL::WiFiClientSecure);
http.begin(std::move(client), uri);
// ‘client’ is solely owned by the HTTPClient now
```

d-a-v, 2021-06-08:
Reopening per https://gitter.im/esp8266/Arduino?at=60bfcfe502f6ee5b91e06200 and per closed unfinished #5739 .
@holgerlembke

holgerlembke, 2021-06-09:
Ok, please help me to understand the problem. First, reuse. I look at the ReuseConnection-Example from ESP8266HTTPClient. 

Where is the connection-resue? The loop() has "WiFiClient client;" so I assume the connection is created everytime.

A valid reuse-example would look like:
```
/**
   reuseConnection.ino

    Created on: 22.11.2015

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

ESP8266WiFiMulti WiFiMulti;

HTTPClient http;
WiFiClient client;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("SSID", "PASSWORD");

  // allow reuse (if server supports it)
  http.setReuse(true);
}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    if (client.connected()) {
      Serial.println("\n!!Connected!!!\n");
    }

    http.begin(client, "http://jigsaw.w3.org/HTTP/connection.html");
    //http.begin(client, "jigsaw.w3.org", 80, "/HTTP/connection.html");

    int httpCode = http.GET();
    if (httpCode > 0) {
      Serial.printf("[HTTP] GET... code: %d\n", httpCode);

      // file found at server
      if (httpCode == HTTP_CODE_OK) {
        http.writeToStream(&Serial);
      }
    } else {
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }

    http.end();
  }

  delay(5000);
}
```
Right?

d-a-v, 2021-06-09:
> Where is the connection-reused?

In the HTTP client class.
The first time connection is made HTTP/1.1, `client` connection is not closed, is kept open between loops by `http` instance.

> The loop() has "WiFiClient client;" so I assume the connection is created everytime.

In your example it is not: it is declared globally. That's how it can be kept opened between loops.

holgerlembke, 2021-06-09:
Really? In begin(client...) _client is assigned the newly created WiFiClient.

d-a-v, 2021-06-09:
> Really? In begin(client...) _client is assigned the newly created WiFiClient.

It is given, not created. What http client does it it under the hood might be different from what you think (now I understand [your question in gitter](https://gitter.im/esp8266/Arduino?at=60c0ca3f5f235f69507e7911)).
`http.begin(client, url)` can be thought of "begin a new request, create the connection if needed, reuse it if possible".

holgerlembke, 2021-06-09:
Some time has passed. Broken example https://github.com/esp8266/Arduino/pull/8111 has been removed.

Now that I know how reuse shall work, there is the situation that reuse is true by default. Let us do an experiment with BasicHttpClient-example :

```
    if (http.begin(client, "http://jigsaw.w3.org/HTTP/connection.html")) {  // HTTP
        .....
        http.end();
    }
    if (http.begin(client, "http://google.com/")) {  // HTTP
          ##.....
        http.end();
    }
```
Obviously (really?) this does not work, because the connection is still to w3.org and we only get something useful by chance. If I set reuse to false I get the result I expect.

This is all quite obviously if you read the source code of HttpClient. But if I had to solve something requesting something from multiple different urls and without reading much source/docu I might have done a looping solution with begin()-end() and not so much have this reuse situation on my radar.

Next experiment: change the construction order:
```
    HTTPClient http;
    WiFiClient client;
    http.setReuse(false);
```
This works fine. Because in end()/disconnect() the connection is closed and the pointer to _client nulled.

Soooooooo. Conclusion:

* Just change the default of reuse to false. It will create a working solution in all situations without any crashes.
* Add an explanation into ESP8266HTTPClient.h at https://github.com/esp8266/Arduino/blob/2185f9bd73cd9a49d7b40f2c088a19b0db611f55/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h#L175 why reuse is good, how and when to use and the pitfalls in destructor calling.

PR #8237 (2021-10-13): Movable HTTPClient and fixing WiFiClient copy
- =default for default ctor, destructor, move ctor and the assignment move
- use std::unique_ptr<WiFiClient> instead of raw pointer
- implement clone() for the WiFiClient so it's safe to copy the WiFiClientSecure, and not accidentally call the WiFiClient copy instead
- replace headers pointer array with unique_ptr<T[]> to simplify the move operations
- substitute userAgent with the default one when it is empty

Allow HTTPClient to be placed inside of movable classes (e.g. std::optional) or to be returned from functions. Class logic stays as-is, onl [... cut]