esp8266/Arduino #5737: Chip Crashing when changing from SoftAP Mode to Station Mode

Aggeloz, 2019-02-08:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			        [ESP-12]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

So i want to make a captive portal on my esp8266 so i can put my WiFi info and then use them to connect to my home wifi....I start with SoftAP mode i serve a page that i just put my WiFi info into then i press a button it sends the info to the chip as a JSON file i parse it then put the SSID and PASS on vars then i close the SoftAP and start a Station Mode with the into that i got but every time i try to send the data the chip crashes immediately after it receives the data and gives me this : 

Exception (9):
epc1=0x40208d46 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4f575357 depc=0x00000000

ctx: cont 
sp: 3ffffd50 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffef0:  3ffef7d4 3ffef418 3ffef450 402081cd  
3fffff00:  00000000 3fffdad0 3ffef9f8 00000030  
3fffff10:  00000000 3fffdad0 3ffe85f8 3ffef418  
3fffff20:  3ffef7d4 000000b4 4020a36c 3fffefb0  
3fffff30:  3ffef7d4 3ffef418 00000000 402017af  
3fffff40:  00000000 3ffef418 3ffef504 40208da8  
3fffff50:  00000000 00000000 4bc6a7f0 00000000  
3fffff60:  4020208d 00000000 40106644 3ffef9f8  
3fffff70:  3fffdad0 00000000 3ffef418 40208dd9  
3fffff80:  00d5f721 00000000 3ffef418 40203bed  
3fffff90:  00000000 00000000 00000000 3ffef9f8  
3fffffa0:  00000000 00000000 3ffef9f2 40203c31  
3fffffb0:  3fffdad0 00000000 3ffef9f2 4020a3f8  
3fffffc0:  feefeffe feefeffe 3ffe85f8 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WebSocketsServer.h>
#include <ArduinoJson.h>
#include <WebSocketClient.h>
#include <string>
#include "DHT.h"
#define DHTPIN 5
[... 972 lines cut ...]
    }
  

}

void webthings(){
  webSocket.loop();
  server.handleClient();
  if(Serial.available() > 0){
    char c[] = {(char)Serial.read()};
    webSocket.broadcastTXT(c, sizeof(c));
  }

  }

  void loop()
{
  webthings();
  unsigned long currentMillis = millis(); //(Millis instead of delays so it's not interupting other parts of the code)
  if(currentMillis - previousMillis >= interval && isconnected == 1){
    previousMillis = currentMillis;
    TEMPsense();

    }
}

ToyboxLabs, 2019-02-19:
I'm having the same problem, what was the solution?

Aggeloz, 2019-02-19:
> I'm having the same problem, what was the solution?

i didn't find one i just closed the issue cause no one replied to it....should i re-open it?

ToyboxLabs, 2019-02-19:
Yeah I am trying to narrow down my code to see exactly what is causing it, but it seems to be some corruption when I switch between AP mode and STA mode.

Aggeloz, 2019-02-19:
> Yeah I am trying to narrow down my code to see exactly what is causing it, but it seems to be some corruption when I switch between AP mode and STA mode.

i think it has to do something with the way im storing the values for the SSID and the PASSWORD for the new wifi but i could be wrong

ToyboxLabs, 2019-02-20:
Mine ended up being a completely different issues, so sorry to confuse you. I would suggest keep removing code until it doesn't repro and it should give you a better idea of what is actually happening.

Aggeloz, 2019-02-20:
> Mine ended up being a completely different issues, so sorry to confuse you. I would suggest keep removing code until it doesn't repro and it should give you a better idea of what is actually happening.

What was your issue i might give your solution a try