esp8266/Arduino #5738: Networking:430:47: error: no matching function for call to 'UdpContext::listen(const ip_addr_t&, int)'

dony71, 2019-02-08:
compile latest espeasy mega release then result this 
compile with v.2.5.0 and arduino ide 1.8.8 
anything i'm missing?

>F:\arduino-1.8.8P250\ESPEasy\Networking.ino: In function 'bool SSDP_begin()':
>
>Networking:430:47: error: no matching function for call to 'UdpContext::listen(const ip_addr_t&, int)'
>
>   if (!_server->listen(*IP_ADDR_ANY, SSDP_PORT)) {
>
>                                               ^
>
>F:\arduino-1.8.8P250\ESPEasy\Networking.ino:430:47: note: candidate is:
>
>In file included from sketch\ESPEasy-Globals.h:515:0,
>
>                 from F:\arduino-1.8.8P250\ESPEasy\ESPEasy.ino:90:
>
>F:\arduino->1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpC>ontext.h:98:10: note: bool UdpContext::listen(const ip_addr_t*, uint16_t)
>
>     bool listen(CONST ip_addr_t* addr, uint16_t port)
>
>          ^
>
>F:\arduino->1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpC>ontext.h:98:10: note:   no known conversion for argument 1 from 'const ip_addr_t {aka const ip4_addr}' >to 'const ip_addr_t* {aka const ip4_addr*}'
>
>Networking:437:50: error: no matching function for call to 'UdpContext::connect(ip_addr_t&, int)'
>
>   if (!_server->connect(multicast_addr, SSDP_PORT)) {
>
>                                                  ^
>
>F:\arduino-1.8.8P250\ESPEasy\Networking.ino:437:50: note: candidate is:
>
>In file included from sketch\ESPEasy-Globals.h:515:0,
>
>                 from F:\arduino-1.8.8P250\ESPEasy\ESPEasy.ino:90:
>
>F:\arduino->1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpC>ontext.h:91:10: note: bool UdpContext::connect(const ip_addr_t*, uint16_t)
>
>     bool connect(const ip_addr_t* addr, uint16_t port)
>
>          ^
>
>F:\arduino->1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpC>ontext.h:91:10: note:   no known conversion for argument 1 from 'ip_addr_t {aka ip4_addr}' to 'const >ip_addr_t* {aka const ip4_addr*}'
>
>Multiple libraries were found for "Servo.h"
> Used: F:\arduino-1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\Servo
> Not used: F:\arduino-1.8.8P250\libraries\Servo
>exit status 1
>no matching function for call to 'UdpContext::listen(const ip_addr_t&, int)'
>
>

d-a-v, 2019-02-08:
Can you try this simple change ?

https://github.com/esp8266/Arduino/pull/5742/files#diff-00d285ad803572cb56bf12e364202cd9R98

AlfredLamoule, 2019-02-10:
Just had the exact same problem, this change solved the problem.

But the same error appeared with the `connect` method.
Using the same logic worked perfectly for me, I was finally able to build the firmware :
``` 
#if LWIP_VERSION_MAJOR == 1
    bool connect(IPAddress addr, uint16_t port)
    {
        _pcb->remote_ip = addr;
        _pcb->remote_port = port;
        return true;
    }
#else
    bool connect(const IPAddress& addr, uint16_t port)
    {
        _pcb->remote_ip = addr;
        _pcb->remote_port = port;
        return true;
    }
#endif
```

d-a-v, 2019-02-11:
@AlfredLamoule I pushed your fix in #5742

PR #5742 (2019-02-11): fix UdpContext::(connect,listen) signature by using IPAddress
fix #5738 
`::connect()` fix by courtesy of @AlfredLamoule
closed 2019-02-11 by d-a-v