esp8266/Arduino #5746: unexpected dhcps lease set to default by wifi_softap_dhcps_start and by wifi_set_opmode

algood, 2019-02-08:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			[**ESP-12**|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date **March 2018 ESP8266-2.4.1**]
- Development Env:		[Arduino IDE|Platformio|**Make**|other]
- Operating System:		[Windows|**Ubuntu**|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|**Wemos D1 mini r2/pro**|Nodemcu|other]
- Flash Mode:			[qio|**dio**|other]
- Flash Size: 		[**16MB**/4MB/1MB]
- lwip Variant:   [**v1.5 open Lwip**|v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[**Wemos button or usb access**|ck|nodemcu]
- Flash Frequency:	[**esptool**|40Mhz]
- CPU Frequency:		[**80Mhz**|160MHz]
- Upload Using:		[OTA|**SERIAL]**
- Upload Speed:   [115200|**921k**] (serial upload only)

### Problem Description

espressif **libmain** is at stakes:
wifi_set_opmode() triggers wifi_softap_dhcpcs_start() with espressif default IP lease in 192.168.4.x. Some stations will quickly reconnect with 192.168.4.x address and not in the IP range we specify usually just after wifi_set_opmode.
Could you remove the wifi_softap_dhcps_start from wifi_set_opmode ?
A better understanding of operations performed in wifi_set_opmode (extract of source code) could help.

As well:
In page 50 of the ESP8266_Non_OS_SDK_API pdf document, it is written that wifi_softap_dhcps_stop/_start initialize lease with espressif default lease. In code following this function, we have to set the lease we want, even it's been done before. And as said above, wifi_softap_dhcps_start is performed in wifi_set_opmode and erase our lease. Could wifi_set_opmode or wifi_softap_dhcps_start avoid modifying the current dhcps lease ?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

NA

### Debug Messages

NA

devyte, 2019-11-09:
@d-a-v is this impacted by your ongoing wip and ethernet changes? Could the dhcp start/stop calls be conditional?

d-a-v, 2019-11-09:
The [ethernet PR](https://github.com/esp8266/Arduino/pull/6680) brings DHCP server right here.
We will have full control over it (we already had it but it will be easier).