esp8266/Arduino #5748: pins.png transfer from sd card to webserver using server.streamFile(), raise the error:Exception (9)

nethunter787, 2019-02-09:
forgive my poor english,I have been stucked in this problem for a long time.(I have tried Google.and seen the esp8266webserver.h)
NodeMCU 1.0 ESP-12Emodule
CPU 80Mhz
flash(4M 1M SPIFF)
IwIP Variant :V2 lower Memory

using MicroSD Card Adapter(5v->VCC)

here is the sketch 
`/*
  SDWebServer - Example WebServer with SD Card backend for esp8266

  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the ESP8266WebServer library for Arduino environment.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA

  Have a FAT Formatted SD Card connected to the SPI port of the ESP8266
  The web root is the SD Card root folder
  File extensions with more than 3 charecters are not supported by the SD Library
  File Names longer than 8 charecters will be truncated by the SD library, so keep filenames shorter
  index.htm is the default index (works on subfolders as well)

  upload the contents of SdRoot to the root of the SDcard and access the editor by going to http://esp8266sd.local/edit

*/
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <SPI.h>
#include <SD.h>

#define DBG_OUTPUT_PORT Serial

#ifndef STASSID
#define STASSID "CMCC_10050_F1DA50"
#define STAPSK  "13762806488"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;
const char* host = "esp8266sd";

ESP8266WebServer server(80);

static bool hasSD = false;
File uploadFile;

void returnOK() {
  server.send(200, "text/plain", "");
}

void returnFail(String msg) {
  server.send(500, "text/plain", msg + "\r\n");
}

bool loadFromSdCard(String path) {
[... 312 lines cut ...]
3ffffe90:  3ffe86bb 3ffee9f8 3ffeea14 3ffeee44  
3ffffea0:  00000000 3ffeea14 3ffeea14 40202910  
3ffffeb0:  3fff013c 0000000f 00000009 00000000  
3ffffec0:  00000000 00000000 00000000 00000001  
3ffffed0:  00000000 3fffc6fc 00000000 3ffeee44  
3ffffee0:  00000000 3ffeea14 3ffee9d0 4020f7a2  
3ffffef0:  00000000 00000000 00000000 4020564a  
3fffff00:  00000000 00000000 00000000 40205772  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 4010014c 00001cf8  
3fffff30:  3ffeea14 00000001 3ffee9d4 4020f778  
3fffff40:  00217fb0 00000000 00000000 0000000f  
3fffff50:  00000000 3fff06e4 3ffee9d0 3ffeee44  
3fffff60:  00000001 3ffee9f8 3ffee9d0 40205964  
3fffff70:  40210140 00000000 00001388 40208ab4  
3fffff80:  00000000 3fff06e4 3ffeee14 402071f6  
3fffff90:  3fffdad0 00000000 3ffeee14 402031fc  
3fffffa0:  3fffdad0 00000000 3ffeee14 4020f2a0  
3fffffb0:  feefeffe feefeffe 3ffe8518 40100d21  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
However.
when I try to replace the pins.png with  a 1 KB flag.png.It works fine and successfully loading it within 10s. 
I'm wondering whether the Function streamFile()  have devided the pins.png into pices to upload? (because it's too large)

d-a-v, 2019-02-11:
after `while (toRead > 0) {` in [libraries/SD/src/utility/SdFile.cpp#L682](https://github.com/esp8266/Arduino/blob/master/libraries/SD/src/utility/SdFile.cpp#L682), can you please try to add this:

```
delay(0);
```
and report ?

earlephilhower, 2019-02-11:
@nethunter787 , please decode the stack dump using the Tools->ESP Exception Decoder, otherwise your report is not useful since we cannot tell where the crash is.

earlephilhower, 2019-02-11:
@nethunter787  You deleted all the information we asked for in the template, so this report is not usable.  Please make a new bug and fill out all sections of the report and decode the stack dump.  This bug doesn't even list the core version or decode the stack.

closed 2019-02-11 by earlephilhower