esp8266/Arduino #5749: 2.5.0: Issues with Arduino API

d-a-v, 2019-02-09:
> UDP `virtual IPAddress remoteIP() =0;` `virtual IPAddress remoteIP() const =0;`
you can't change a virtual function in an established Arduino core class

https://github.com/esp8266/Arduino/commit/5c4db3acf4353870b6a5663ba06d712d67e6c179#commitcomment-32254962

> other networking libraries Client implementations do not implement `flush` and `stop` with parameter.
"The type 'UIPClient' must implement the inherited pure virtual method 'Client::stop' "

https://github.com/esp8266/Arduino/commit/83a8076db87b77de6a7b500c18b85057e38ed117#commitcomment-32254941

@jandrassy Can you please give more details ?

Also [If your sketch does not compile and returns error](https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-arduino-ide) needs to be addressed.

jandrassy, 2019-02-10:
There are other networking libraries used with esp8266, not only the native WiFi or bundled Ethernet library (why is it there?). If you change the virtual functions of the Arduino networking base classes Client, Server, UDP the libraries don't compile. Function `flush` with parameter is a different virtual function then `flush` without parameter. The default value of the parameter doesn't change it. So now  for example the UIPEthernet library or the standard Ethernet library and perhaps some GPRS libraries don't compile for esp8266 because they don't implement flush, stop, remoteIP with a parameter.

d-a-v, 2019-02-20:
It will be addressed for next release 2.5.1

PaulStoffregen, 2019-02-20:
FWIW, there was an unresolved problem with DNS lookups by the Ethernet lib.  Only ESP8266 has this problem.  The exact same code ran perfectly on all other boards, including ESP32.  I never did a deeper investigation into why that failed on ESP8266.  Maybe someday....

If anyone cares about Wiznet-based Ethernet on ESP8266, please test some of the library's examples which use a text-based hostname rather than only IP numbers.

jandrassy, 2019-02-26:
recommended
https://github.com/arduino/ArduinoCore-API

jandrassy, 2019-04-14:
> FWIW, there was an unresolved problem with DNS lookups by the Ethernet lib. Only ESP8266 has this problem. The exact same code ran perfectly on all other boards, including ESP32. I never did a [...]

@PaulStoffregen, I tested it now and both examples with DNS work with esp8266 core 2.4.2 (hw: Wemos D1 mini clone with WIZ850io)

@d-a-v, I think the bundled Ethernet library should be removed from esp8266 arduino boards package

d-a-v, 2019-04-14:
> I think the bundled Ethernet library should be removed from esp8266 arduino boards package

That can be done after #5969 is merged.
We also can replace it by a submodule linking to @PaulStoffregen repository.
This is not mandatory since we have no examples external to this library using Ethernet class.
Comments about that are welcome.

jandrassy, 2019-04-15:
> That can be done after #5969 is merged.
> We also can replace it by a submodule linking to @PaulStoffregen repository.
> This is not mandatory since we have no examples external to this library using [...]

bundle only architecture specific libraries. for the Ethernet library the SPI library does the abstraction

d-a-v, 2019-04-22:
Did you need to make changes to the new Ethernet library ? I had to (https://github.com/arduino-libraries/Ethernet/pull/96).

> (@jandrassy) I tested it now and both examples with DNS work with esp8266 core 2.4.2 (hw: Wemos D1 mini clone with WIZ850io)

#5997 removes old ethernet library and replace it with a submodule. Would you be able to test it ?

I currently can't make it pointing to Ethernet repo so I made a fork. It would be far better to point to master when the Ethernet PR is merged, especially for release 2.5.1 (or I'd need to keep this fork forever on my repo).

jandrassy, 2019-04-23:
> Did you need to make changes to the new Ethernet library ? I had to ([arduino-libraries/Ethernet#96](https://github.com/arduino-libraries/Ethernet/pull/96)).
 
I was on 2.4.2

PR #5969 (2019-04-26): restore proper arduino Client:: & Wire:: API
fixes #5749
fixes #5901 
fixes #5705 
~includes #5997~
closed 2019-04-26 by d-a-v