esp8266/Arduino #5753: espconn.h compile error ipv4_addr_t undefined

martin-ger, 2019-02-11:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[V2.5, 11/2/2019]
- Development Env:		[Arduino IDE]
- Operating System:		[UbuntuS]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Include of espconn.h results in an compiler error (worked in V2.4.2):
```
/home/mfg/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/include/espconn.h:451:54: note: in expansion of macro 'ipv4_addr_t'
 typedef void (*dns_found_callback)(const char *name, ipv4_addr_t *ipaddr, void *callback_arg);
                                                      ^
WiFiClient:2:21: error: 'ip_addr_t' has not been declared
```
Following addition in espconn.h solves it:
```cpp
#ifndef ipv4_addr_t
#define ipv4_addr_t ip_addr_t
#endif
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <espconn.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

pfeerick, 2019-02-19:
Are you not including anything else, as if you  `#include <ESP8266WiFi.h>` or `#include <IPAddress.h>` (I can't see what's filling in the ipv4_addr_t though) before `espconn.h`, for instance, it will compile just fine with LWiP 1.4 HB... 

Is espconn supposed to work with LWIP1.4 though? ... as the [commit that added espconn](https://github.com/esp8266/Arduino/commit/0db6ec4ba831180e50275f6b3cb236b13b84d395) states "lwip2: port esp-ping and espconn"

devyte, 2019-11-09:
@d-a-v I suspect this can be closed. Do we still support the espconn layer?

d-a-v, 2019-11-09:
espconn is supported by lwIP-1.4
it is not officially supported by lwip2, but it is ported and untested.
`espconn_init()` has to be called before using it with lwip2.

@pfeerick do you intend to make a PR with your proposed update ?

pfeerick, 2019-11-09:
@d-a-v Probably should do... completely forgot about this as the OP never replied.

PR #6740 (2019-11-09): Add `ipv4_addr_t` for espconn/LwIP1.4 compat
`espconnn.h` references ipv4_addr_t, so it seems appropriate it be defined here also.

If `#include <IPAddress.h>` added to MCVE, closes #5753. 

The change to `espconn.h` suggested by the OP

```
#ifndef ipv4_addr_t
#define ipv4_addr_t ip_addr_t
#endif
```

sets off needing to define `ip_addr_t`, etc, so is not suitable.
closed 2019-11-09 by d-a-v