esp8266/Arduino #5755: ESP8266 boot stuck if TX/RX are connected.

chewax, 2019-02-12:
This is the thing. I have configured my RX/TX pins to be used as input (INPUT_PULLUP to be more specific). Im building a standalone device so it must be really robust (specially to power cuts, since they are pretty frequent). The problem arises when the board is forced to reboot (either pulling the RST pin or flicking the power). It gets stuck on the boot (i have isolated the problem to the RX/TX pins). If those are connected to circuit on boot, then the whole boot is halted. If i disconnect, boot and then connect, everything works perfect. So here is the question. Is there a way to inform the board not to pay attention to those pins on boot or so? Any help is well appreciated! Thanks!

igrr, 2019-02-12:
Not by software means — the bootloader in the mask ROM checks the state of bootstrapping pins, so application can not affect its behavior — it's not loaded yet. You can probably work around this using some external circuit, but such discussion would be out of scope of this repository.

chewax, 2019-02-12:
Ok thanks a lot. This is exactly what I thought and was hoping I was wrong...well guess I will have to redesign everything. Thanks again!

closed 2019-02-12 by chewax