esp8266/Arduino #5756: ESP8266 Hardware watchdog reset

martinalberto, 2019-02-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[]
- Flash Size: 		[4MB]
- lwip Variant:   []
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
Once I get a reset of Guardian Dog type Hardware.

In these cases, you give me the following Debuger information:

>   ets Jan 8 2013, rst cause: 2, boot mode: (1,7)
> 
> 
>   ets Jan 8 2013, rst cause: 4, boot mode: (1,7)
> 
> wdt reset

And the team remains frozen. The Reset does not continue. The program does not start again.

What worries me is that when the guard dog jumps it gets stuck.

I understand that this is due to the starting address of the memory.

How can we fix it?
I think the Watchdog should reset and start the program again when it jumps. and DO NOT leave the equipment frozen.

igrr, 2019-02-12:
`boot mode: (1,7)` indicates that GPIO0 was low on reset, so the chip boots into UART download mode, not into your application.

isnadh, 2019-02-13:
Check your hardware, something is keeping GPIO0 low on boot.

martinalberto, 2019-02-13:
Ok, it is the problem. 
I usually use the pin GPIO0

Thanks !

closed 2019-02-13 by martinalberto