esp8266/Arduino #5761: Deep sleep forever and wake-up on interrupt

Erik84750, 2019-02-15:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:				ESP12-F development board
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 7 64 bit]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[4MB 1MB SPIFFS]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [256000)

### Problem Description

How to implement Deep Sleep Forever with external interrup active low (from DS3231 output set as alarm timer)

Can the ESP8266 be brought to "deep sleep forever" with wake-up on the RST pin from active-low output from DS3231?
What code is needed to bring the ESP8266 in deep-sleep-mode-forever with wake-up on the RST pin (using active low pulse from DS3231)?

duncan-a, 2019-02-15:
Depending upon your application, you might want to consider using a TPL5110
to completely power down the ESP (and drop quiescent current down to nA)
and then power it up periodically.

Erik84750, 2019-02-15:
I know the TLP5110, disadvantage is that its clock is not precise and its offset from the real time accumulates. An RTC module (DS3231 for example) will give an exact wake up time correct to within seconds even after one year of operation.

duncan-a, 2019-02-15:
I agree that the TLP5010 is only as accurate as its timing resistors and
there can be thermal drift. If the 'done' pulse is implemented, the next
switchon is at the set time minus the used time, so the starts should
always be consistent (drift excepted).

Maybe you could latch the RTC alarm pulse, use that to fire a MOSFET on the
power line of the ESP and issue a reset pulse to the latch when the ESPs
code completes.

Just a thought (I don't have a lot of faith in Deep Sleep).

Erik84750, 2019-02-15:
ok, so basically no need for any low power library nor interrupts: get the ESP8266 in permanent deep sleep, and reset RST pin to wake up..

Why no faith in ESP8266 deep sleep?

duncan-a, 2019-02-15:
No, I mean just have the ESP power up, run the code and power itself down
again. Wouldn't be quite as low power as the 5110, because of the latch,
but it may be better than Deep Sleep.

Basically, I've never got Deep Sleep to behave as I expected it to...

Erik84750, 2019-02-15:
The issue in my case is that I need the ESP8266 to perform a task at exactly xx:yy, every time, the same time, once an hour

duncan-a, 2019-02-15:
What I'm suggesting would achieve that. The ESP is pretty consistent
time-wise when running code from a power-on bootup.

Erik84750, 2019-02-15:
Using the TLP5110 the wake-up of the ESP8266 is dependent on the timing accuracy of this chip, which is using a RC timer: not exactly the accuracy required in my initial post.

devyte, 2019-02-15:
This is a "how do I..." type question. As explained in the issue template, please seek such help at a community forum like esp8266.com or stackoverflow.

Closing.

closed 2019-02-15 by devyte

duncan-a, 2019-02-15:
At the risk of upsetting Big Brother (correct though he may be), I meant
that the RTC through a latch would provide the repeatability you need and
you would have the accuracy of the RTC.