esp8266/Arduino #5767: EEPROM write/put not checking every single byte

mamama1, 2019-02-15:
Hi

it seems to me, that EEPROM.commit(), write and put don't check every single byte if it has changed (using the _dirty flag) but as soon as one byte gets changed, the whole area will be rewritten to flash.

Is this intended or was it some getting-things-done-whoopsie?

eg. I change just one byte but I have reserved 4k of EEPROM, all 4k will be erased and rewritten, if I understand the code correctly.

Thanks

penfold42, 2019-02-15:
This is intended behaviour.
It’s an emulation layer on top of a SPI flash sector and SPI flash sectors must be erased before writing.

mamama1, 2019-02-15:
> It’s an emulation layer on top of a SPI flash sector and SPI flash sectors must be erased before writing.

I know that, but how big is one sector? 4k?

devyte, 2019-02-15:
Correct. Closing as intended behavior.

closed 2019-02-15 by devyte