esp8266/Arduino #5773: strange interaction between the Ticker and WiFiClientSecure libraries

vlad2005, 2019-02-17:
Hi!
I use platformio with esp8266. This is my current setup:
Platform Espressif 8266
--------
Updating espressif8266                   @ 1.8.0          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20402.4      [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

I have a complex application, but basically it is about the following situation:
I have an function named "getStatus" which connects to googleapis to get some data from firestore. This function use HTTPSRedirect library (i will change that in future), but I used as replacement for WiFiClientSecure library. Anyway, I don't have in googleapis any redirection.
In this function i have normal sequences for init client, make connection and then make an POST request to get data.
Function _getStatus_ is called in two ways
1. from interaction with an web page on esp8266. In this case everything works as it should
2. using Ticker library to call function at some time interval. In this case everytime a get error at sequence where client try to connect to googleapis ( client->connect(host, httpsPort).)
I replaced Ticker library with direct use of ESPTimer, os_timer_setfn, os_arm, .... and so on, but i have same problem which was somehow expected.
My guess is that there is a kind of conflict between using the timer in esp8266 and implementing for WiFiClientSecure.
Unfortunately, I do not have time to do a lot of tests because I have another job. Programming is hobby for me. And sorry for me english.
If is not clear what i want to say, i will tray in next days to make an simple app for test this behavior.

bertmelis, 2019-02-17:
Ticker callbacks are interrupt driven. So you can not do lengthy operations. Just set a flag and do the main work in the Arduino loop.

vlad2005, 2019-02-17:
OMG!!!!
I forgot about this. In fact, I have taken this aspect into account for a long time. But recently we made some changes, and we also dropped this implementation with the intention of simplifying the application. Now is clear. Pfufff.

Thanks!
This issue can be marked as solved.

closed 2019-02-17 by devyte