esp8266/Arduino #5775: Exception 9 & NTP - Bad coding, flaky exceptions or core issue?

pfeerick, 2019-02-18:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[Digistump Oak]
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Digistump Oak]
- Flash Size: 		        [4MB]
- lwip Variant:                   [v2 Lower Memory]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [230400]

### Problem Description

In the process of trying to figure out what is causing another project to randomly crash, I thought I'd start going back to basics, and throw out old code. First up was NTP - I wanted to use the internal stuff since I found out it's there in the core. I came across this version first, and basically just shoved in OTA and WifiManager to configure it. It initially worked for a few minutes, and then started freezing. If I remove the `if (tnow % 60)` re-direct, and let it output every second, it appears to remain stable. I then found the example included with the core, and it seems to be working just fine on the same hardware. 

I was going to ignore this, but when I tried enabling Exceptions, the code didn't just crash... but froze... making me wonder if it is just something wrong with this combination, or something else weird. I know the coding is currently in an atrocious kludged state... but I was just wondering if it's indicative of a deeper problem, or just something that the NTP-TZ-DST example code has subsequently fixed?

Am I right in thinking the  
```
ets Jan  8 2013,rst cause:2, boot mode:(1,7)
ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

combo is indicative that there is too long a delay happening somehow, and triggering the s/w watchdog? Although having said that... why does the ESP8266 not restart after `wdt reset`? It's just sitting there dead turkey until I power cycle it.  I'll see how I go getting rid of more of the code to see what the MCVE really is in the next day or two...

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h> // For OTA programming

#include <time.h>                       // time() ctime()
#include <sys/time.h>                   // struct timeval
#include <coredecls.h>                  // settimeofday_cb()

#define OAK_NAME "oak3"

////////////////////////////////////////////////////////

timeval cbtime;			// when time set callback was called
bool cbtime_set = false;

// callback function that is called whenever time is set
void time_is_set (void)
{
  time_t t = time(nullptr);

  gettimeofday(&cbtime, NULL);
  cbtime_set = true;
  Serial.println("------------------ settimeofday() was called ------------------");
  printf(" local asctime: %s\n", asctime(localtime(&t))); // print formated local time
}

void configModeCallback(WiFiManager *myWiFiManager)
{
  Serial.println(WiFi.softAPIP());
  Serial.println(myWiFiManager->getConfigPortalSSID());
}

void setup() {
  Serial.begin(115200);

  settimeofday_cb(time_is_set); // optional: set function to call when time is set

  configTime(0, 0, "pool.ntp.org");

  // set up TZ string to use a POSIX/gnu TZ string for local timezone
  setenv("TZ", "AEST-10", 1);

  tzset(); // save the TZ variable

  WiFiManager wifiManager;

  wifiManager.setAPCallback(configModeCallback);

  String autoconf_ssid = OAK_NAME + String(ESP.getChipId());

  if (!wifiManager.autoConnect())
  {
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(1000);
  }

  ArduinoOTA.setHostname(OAK_NAME);
  ArduinoOTA.begin();

  Serial.println();

}

// for testing purpose:
extern "C" int clock_gettime(clockid_t unused, struct timespec *tp);

#define PTM(w) \
  Serial.print(":" #w "="); \
  Serial.print(tm->tm_##w);

void printTm (const char* what, const tm* tm)
{
  Serial.print(what);
  PTM(isdst); PTM(yday); PTM(wday);
  PTM(year);  PTM(mon);  PTM(mday);
  PTM(hour);  PTM(min);  PTM(sec);
}

timeval tv;
struct timezone tz;
timespec tp;
time_t tnow;

void loop()
{
  ArduinoOTA.handle();
  gettimeofday(&tv, &tz);
  clock_gettime(0, &tp); // also supported by esp8266 code
  tnow = time(nullptr);

  // localtime / gmtime every second change
  static time_t lastv = 0;
  if (lastv != tv.tv_sec)
  {
    lastv = tv.tv_sec;

    if (tnow % 60)
       goto noprint;
       
#if 1
    printf("tz_minuteswest: %d, tz_dsttime: %d, millis: %d\n", tz.tz_minuteswest, tz.tz_dsttime,millis());
    printf("gettimeofday() tv.tv_sec : %ld\n", lastv);
    printf("time()            time_t : %ld\n", tnow);
    Serial.println();
#endif

    printf("         ctime: %s", ctime(&tnow)); // print formated local time
    printf(" local asctime: %s", asctime(localtime(&tnow))); // print formated local time
    printf("gmtime asctime: %s", asctime(gmtime(&tnow))); // print formated gm time

    // print gmtime and localtime tm members
    printTm("      gmtime", gmtime(&tnow));
    Serial.println();
    printTm("   localtime", localtime(&tnow));
    Serial.println();

    Serial.println();
  }
  noprint:
  delay(100);
}

```

### Debug Messages

```
tz_minuteswest: 0, tz_dsttime: 0, millis: 101587
gettimeofday() tv.tv_sec : 1550487721
time()            time_t : 1550487720

         ctime: Mon Feb 18 21:02:00 2019
 local asctime: Mon Feb 18 21:02:00 2019
gmtime asctime: Mon Feb 18 11:02:00 2019
      gmtime:isdst=0:yday=48:wday=1:year=119:mon=1:mday=18:hour=11:min=2:sec=0
   localtime:isdst=0:yday=48:wday=1:year=119:mon=1:mday=18:hour=21:min=2:sec=0

Exception (9):
epc1=0x40246a12 epc2=0x00000000 epc3=0x00000000 excvaddr=0xfff00123 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffffcb0 end: 3fffffb0 offset: 01a0
3ffffe50:  00000001 00000000 00000000 4024174b
3ffffe60:  00003731 00000000 00000014 4024561c
3ffffe70:  3ffee8b0 40245631 3ffee8b0 00000000
3ffffe80:  3ffeea02 3ffee8c0 00000001 00000003
3ffffe90:  3ffee8b0 60000600 60000200 3ffee8b0
3ffffea0:  4024130f 00000000 00000000 3ffeea02
3ffffeb0:  40105faa 60000e00 00000000 00000001
3ffffec0:  40246338 3ffe8ee4 00000000 3fffdab0
3ffffed0:  00000000 3fffdcb0 3ffee400 00000008
3ffffee0:  00000000 078a13de 00000000 402495d9
3ffffef0:  00000008 00000000 f7ffffff 60000600
3fffff00:  00000001 00000000 00000000 4021d265
3fffff10:  40239218 00000008 3ffee3e0 078a12c6
3fffff20:  40239642 0789e393 60000600 4021baa6
3fffff30:  078a0fd6 3ffee3e0 40239707 3ffed43c
3fffff40:  3ffee3e0 02029631 3ffee3e0 60000600
3fffff50:  4023d0d5 3ffed43c 3ffee3e0 078a0fd6
3fffff60:  4023d11a 3fffdab0 00000000 3fffdcb0
3fffff70:  3ffee408 40000f65 3fffdab0 00000000
3fffff80:  40000f49 3ffe91e0 3fffdab0 40000f49
3fffff90:  40000e19 40001878 00000002 00000000
3fffffa0:  3fffff10 aa55aa55 000000f0 40104934
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```
Stack trace
```Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40246a12
EXCVADDR: 0xfff00123

Decoding stack results
0x4021d265: sys_timeout_abs at core/timeouts.c line 189
0x4021baa6: memp_free at core/memp.c line 447

```

devyte, 2019-02-19:
After flashing over serial, you have to physically reset the board. Not doing that causes wdt reset when trying to OTA. The wdt can also manifest in other weird ways.
Are you doing that physical reset after serial flashing?

pfeerick, 2019-02-19:
I suspect not consistently (especially as I was mixing it with OTA when the board/IDE was OTA programming). I thought there was a change from soft reset to hard reset along the way to make that no longer necessary... or is that with the python esptool? Or am I talking complete rubbish? :open_mouth: 

Anyway... power cycling removed the double reset and freeze... so the board is now resetting when it b0rks... and has given two different exceptions now... 0 then 9... and then behaved itself so far for a good 45 or so minutes... nearly double the time it took for these exceptions to occur... 

 millis: 1568449 (26 mins)
```
Exception (0):
epc1=0x40246472 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffffd30 end: 3fffffb0 offset: 01a0
3ffffed0:  00000000 3fffdcb0 3ffee408 00000008
3ffffee0:  00000000 6078abcb 00000000 402495d9
3ffffef0:  00000008 00000000 f7ffffff 60000600
3fffff00:  00000001 00000000 00000000 00000000
3fffff10:  40239218 00000008 3ffee3e0 6078aab0
3fffff20:  40239642 1e259061 60000600 4021baa6
3fffff30:  60789c10 3ffee3e0 40239707 3ffed43c
3fffff40:  3ffee3e0 1e2594f3 3ffee3e0 60000600
3fffff50:  4023d0d5 3ffed43c 3ffee3e0 60789c10
3fffff60:  4023d11a 3fffdab0 00000000 3fffdcb0
3fffff70:  3ffee3f0 3fffdad0 3ffeec2c 402088cb
3fffff80:  40000f49 3ffe9120 3fffdab0 40000f49
3fffff90:  40000e19 40001878 00000002 00000000
3fffffa0:  3fffff10 aa55aa55 000000f0 40104934
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbff3a6d9
~ld
*WM:
*WM: AutoConnect
```

```
Exception 0: Illegal instruction
PC: 0x40246472
EXCVADDR: 0x00000000

Decoding stack results
0x4021baa6: memp_free at core/memp.c line 447
0x402088cb: loop_task(ETSEvent*) at /home/pfeerick/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 133
```

millis: 1329846 (22 mins)
```
Exception (9):
epc1=0x40249569 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000c087 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3ffffd70 end: 3fffffb0 offset: 01a0
3fffff10:  40239218 00000008 3ffee3e0 4f48def4
3fffff20:  40239642 18c69399 60000600 4021baa6
3fffff30:  00000000 00000000 40239707 3ffed43c
3fffff40:  3ffee3e0 18c6a541 3ffee3e0 402014e4
3fffff50:  4023d0d5 3ffed43c 3ffee3e0 4f48d458
3fffff60:  4023d11a 3fffdab0 00000000 3fffdcb0
3fffff70:  3ffee3f0 3fffdad0 3ffeec2c 402088cb
3fffff80:  40000f49 3ffe9198 3fffdab0 40000f49
3fffff90:  40000e19 40001878 00000002 3fffffb0
3fffffa0:  3fffff10 aa55aa55 000000f0 40104934
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbff3a6d9
~ld
*WM:
*WM: AutoConnect
```

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40249569
EXCVADDR: 0x0000c087

Decoding stack results
0x4021baa6: memp_free at core/memp.c line 447
0x402014e4: delay_end at /home/pfeerick/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 41
0x402088cb: loop_task(ETSEvent*) at /home/pfeerick/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 133
```

TD-er, 2019-03-06:
What happens if you remove these two lines from the callback function `time_is_set` :
```c++
  Serial.println("------------------ settimeofday() was called ------------------");
  printf(" local asctime: %s\n", asctime(localtime(&t))); // print formated local time
```

devyte, 2019-10-31:
@pfeerick several stability issues have gone in, and at least one could be very relevant. Is this still reproducible with latest git?

pfeerick, 2019-12-01:
I have just revisited this, and I suspect a combination of different development environment, different library versions, and fixes in the core have rendered me unable to reproduce this issue, so I think it can be safely closed. I _was_ using the Arduino IDE, but am now using PlatformIO, and am unable to reproduce this again after 15+ hours runtime on the 2.6.1 (PIO 2.3.1) core, and about 5 hours on the 2.5.0(PIO 2.0.4) that I was using at the time. One less issue for now! :)

closed 2019-12-01 by pfeerick