esp8266/Arduino #5776: fatal error: Adafruit_Sensor.h: No such file or directory

Nayakssn1992, 2019-02-18:
How to solve this problems please guide  #me.
**In file included from C:\Users\MAA\Documents\Arduino\libraries\DHT_sensor_library\DHT_U.cpp:22:0:

C:\Users\MAA\Documents\Arduino\libraries\DHT_sensor_library\DHT_U.h:25:29: fatal error: Adafruit_Sensor.h: No such file or directory

 #include <Adafruit_Sensor.h>

                             ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).**

supersjimmie, 2019-02-18:
https://github.com/adafruit/DHT-sensor-library
Says:
You must have the following Arduino libraries installed to use this class:
Adafruit Unified Sensor Library

devyte, 2019-02-18:
Closing due to user error.

closed 2019-02-18 by devyte

Nayakssn1992, 2019-02-19:
Sir,T_sensor_library also I'm getting errorsplease help me on this.____________________________________________________________________________________________________Arduino: 1.8.7 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"
Multiple libraries were found for "DHT.h"In file included from C:\Users\MAA\Documents\Arduino\libraries\DHT_sensor_library\DHT_U.cpp:22:0:
C:\Users\MAA\Documents\Arduino\libraries\DHT_sensor_library\DHT_U.h:25:29: fatal error: Adafruit_Sensor.h: No such file or directory
 #include <Adafruit_Sensor.h>
                             ^
compilation terminated.
 Used: C:\Users\MAA\Documents\Arduino\libraries\DHT_sensor_library Not used: C:\Program Files\Arduino\libraries\DHT_sensor_libraryexit status 1Error compiling for board NodeMCU 1.0 (ESP-12E Module).
This report would have more information with"Show verbose output during compilation"option enabled in File -> Preferences.

 
 
https://github.com/adafruit/DHT-sensor-library