esp8266/Arduino #5777: toolchain-xtensa @ 1.40802.190218 update breaks builds

TD-er, 2019-02-18:
There has apparently just been an update to the toolchain and now all builds fail.
See for an example this [Travis output](https://travis-ci.org/letscontrolit/ESPEasy/builds/494892720?utm_source=github_status&utm_medium=notification)

How do we revert this update, since now I cannot build anything anymore on most of my nodes (an update is called in the build script which should only update the env which uses stage)
So development has now stopped for me today if this cannot be reverted easily :(

TD-er, 2019-02-18:
Seems to be related to this merge: https://github.com/esp8266/Arduino/commit/9790e1cb7cfa01ac74056fb835e6282af123aba0

d-a-v, 2019-02-18:
I am currently facing in our CI, and I don't think it comes from us.
Maybe @ivankravets has more informations ?

ivankravets, 2019-02-18:
I've just updated all toolchains to resolve https://github.com/platformio/platform-espressif8266/issues/124

Let me fix ASAP.

ivankravets, 2019-02-18:
The issue was linked with "fixed" symbolic links in toolchains. So, I've just made them relative. Please re-start CI build or re-run `$ pio update`.

Sorry for the issue.

closed 2019-02-18 by ivankravets

TD-er, 2019-02-18:
Thanks for the quick fix.