esp8266/Arduino #5778: error: {build.flash_flags}: No such file or directory

thangktran, 2019-02-18:
I tried to integrate travis-ci into my project.
.travis.yml
```
language: c
before_install:
  - "/sbin/start-stop-daemon --start --quiet --pidfile /tmp/custom_xvfb_1.pid --make-pidfile --background --exec /usr/bin/Xvfb -- :1 -ac -screen 0 1280x1024x16"
  - sleep 3
  - export DISPLAY=:1.0
  - wget http://downloads.arduino.cc/arduino-${IDE_VERSION}-linux64.tar.xz
  - tar xf arduino-${IDE_VERSION}-linux64.tar.xz
  - sudo mv arduino-${IDE_VERSION} /usr/local/share/arduino
  - sudo ln -s /usr/local/share/arduino/arduino /usr/local/bin/arduino
install:
  - ln -s $PWD/libraries/aREST-master /usr/local/share/arduino/libraries/aREST-master
  - ln -s $PWD/libraries/NTPClient /usr/local/share/arduino/libraries/NTPClient
  - ln -s $PWD/libraries/AutomatedIrrigatingSystem-MegaOnly /usr/local/share/arduino/libraries/AutomatedIrrigatingSystem-MegaOnly
  - ln -s $PWD/libraries/AutomatedIrrigatingSystem-Shared /usr/local/share/arduino/libraries/AutomatedIrrigatingSystem-Shared
  - arduino --pref "boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-prefs
  - arduino --install-boards esp8266:esp8266
  - arduino --pref "compiler.warning_level=all" --save-prefs
script:
  - arduino --verbose --verify --board ${BOARD} ${SOURCE}
notifications:
  email:
    on_success: change
    on_failure: change

env:
  global:
    - IDE_VERSION=1.8.8
  matrix:
    - BOARD="arduino:avr:mega:cpu=atmega2560" SOURCE="$PWD/srcs/ArduinoMega/ArduinoMega.ino"
    - BOARD="arduino:avr:mega:cpu=atmega2560" SOURCE="$PWD/srcs/ArduinoMega-SaveSchedules/ArduinoMega-SaveSchedules.ino"
    - BOARD="esp8266:esp8266:generic" SOURCE="$PWD/srcs/ESP8266/ESP8266.ino"
```

During verify the source code, i got this error `xtensa-lx106-elf-g++: error: {build.flash_flags}: No such file or directory`.
Could it be that  ESP8266 package?

closed 2019-02-23 by thangktran

chris24gutsch, 2019-03-23:
Hello @thangktran, I'm having this exact same problem. How did you solve your issue?

thangktran, 2019-03-23:
Hello @chris24gutsch,
I checked the build log from Arduino IDE and found out that "esp8266:esp8266:generic" since lots of other parameters are missing. In my case i use: "esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=512K,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200".

chris24gutsch, 2019-03-23:
@thangktran, 
That fixed it. You are an absolute life saver and I cannot thank you enough!