esp8266/Arduino #5781: Finalize callback list implementation, and migrate wifi events to use it.

devyte, 2019-02-19:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					all
- Core Version:			2.5.0
- Development Env:		all
- Operating System:		all

### Settings in IDE

- Module:  			all

### Problem Description

After some discussion, @hreintke pulled out the wifi callback event list implementation into a proposed standalone implementation. That implementation needs to be tested, covered by tests, examples, and the wifi event system needs to be migrated to use it. Once that is done, other callback systems could potentially make use of it.

Ref: #5710

hreintke, 2019-02-20:
@devyte : I can work on this.  

What is your idea on the way forward for this. 
I would propose to first focus on functionality and later on the implementation details (list/vector/move). 

There is a simple example included in the initial PR. I'll do some first tests on memory usage/leaks.
What other tests and examples are you thinking of ?

When these work OK, I'll make a start with wifi event migration.

devyte, 2019-02-20:
@hreintke much appreciated!
What I had in mind was along the lines of:
1. Make sure functionality needed by the wifi event system isn't missing
2. Check api (parameter passing, const refs, typedefs, usability in general
3. Look for other use cases in the core and in our libraries, and check whether additional functionality is needed in any such use cases found
4. Implement simple usage examples based on the use case found in the previous point

hreintke, 2019-02-20:
@devyte :  OK.

Please let me know of possible usecases you know in core/libraries. it's not easy to search for those
What is the expected timeline for 2.6.0. ?

devyte, 2019-11-05:
Merged the callbacklist as experimental, now pending migration of the WiFi callbacks to use it. Pushing back for that second part.

devyte, 2020-07-06:
This is a nice-to-have, so pushing back again pending some time window that allows implementing the code unification.