esp8266/Arduino #5782: WIFI_AP_STA mode not working with core 2.5.0

ikin77, 2019-02-19:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			ESP-12 / ESP-01 / NodeMCU
- Core Version:			2.5.0 (fails) vs 2.3.0 (works)
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  		Generic ESP8266 Module / Nodemcu
- Flash Size: 		3MB/1MB
- Flash Frequency:	40Mhz
- CPU Frequency:	160MHz
- Upload Using:	SERIAL

### Problem Description

Consider the provided sketch, which is almost a copy of one of the providen samples. It works with cores 
2.3.0 and 2.5.0. In both cases you can get a WIFI connection to the SoftAP SSID and check the response:

$ curl http://192.168.4.1
wifi({"ssid":"","connected":"false","ip":"0.0.0.0"});

Now, uncomment the two lines bellow comment "_Enable WIFI_AP_STA mode_". With core 2.3.0 you can get connected again to the SoftAP SSID and check connection status:

curl http://192.168.4.1
wifi({"ssid":"MyRouterSSID","connected":"true","ip":"192.168.120.76"});

With core 2.5.0, there's no way to get a WiFi connection to the SoftAP SSID.

Thanks in advance and best regards.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

#define APSSID "MY-SOFTAP"

/* Set these to your desired credentials. */
const char *ssid = APSSID;

ESP8266WebServer server(80);

/* Just a little test message.  Go to http://192.168.4.1 in a web browser
   connected to this access point to see it.
*/
void handleRoot() {

  char* json = (char*) malloc(100);

  IPAddress ip = WiFi.localIP();
  if (!ip) {
     IPAddress ip(0, 0, 0, 0);
  }

  sprintf(json, "wifi({\"ssid\":\"%s\",\"connected\":\"%s\",\"ip\":\"%d.%d.%d.%d\"});",
          WiFi.SSID().c_str(),
          WiFi.status() == WL_CONNECTED ? "true" : "false",
          ip[0], ip[1], ip[2], ip[3]);
                  

  server.send (200, "application/json", json);

  free(json);
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Configuring access point...");
  /* You can remove the password parameter if you want the AP to be open. */
  
  // Enable WIFI_AP_STA mode
  // WiFi.mode(WIFI_AP_STA);
  // WiFi.begin("MyRouterSSID", "myrouterpass");
  
  WiFi.softAP(ssid);
  

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.on("/", handleRoot);
  server.begin();
  Serial.println("HTTP server started");

}

```

d-a-v, 2019-02-19:
Is it with both lwIP-1.4 and lwIP-2 ?

ikin77, 2019-02-19:
Hi, d-a-v. I don't know how to answer whether the problem is with lwIP-1.4 or lwIP-2. Is there an option to chose lwIP version when uploading a sketch? Is there a default version for core 2.3.0 and core 2.5.0? How can I check what lwIP version I'm using?

Thanks in advance and best regards.

ikin77, 2019-02-20:
Hi. I've testing and it looks like the problem remains with core 2.5.0, no matter whether you use liwp 1.4 or 2.

With core 2.3.0 same sketch works flawlessly.

Thanks again and best regards.

devyte, 2020-02-02:
@ikin77 is this issue still valid with 2.6.3? I use all wifi modes and can change freely between them,.

earlephilhower, 2020-07-29:
Closing as no response for 6 months.  We're quite a few releases ahead of this issue.  If it happens w/2.7.3, please open a new one.

closed 2020-07-29 by earlephilhower