esp8266/Arduino #5786: HTTP SSL client hangs when an access point goes down

AlexeySofree, 2019-02-19:
#### Platform

- Hardware:					[NodeMCU]
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS|Debian]

### Settings in IDE

- Module:  			[Nodemcu (ESP-12E)]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600]

### Problem Description

I see a constant Soft WDT reset/wdt reset when I am using HTTP SSL client and I turn off my access point. 
Below code can work hours, but if I turn my access point, then ESP8266 hangs.

The issue is not reproducible if uncomment the 6 line:
```cpp
bool SSLFlag = false;
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecureBearSSL.h>

bool SSLFlag = true;
//bool SSLFlag = false;

char urlSSL[] = "https://35.237.110.54/test/";
char SSLFingerprint[] = "26 24 FA 2B 26 1D DD 1F 2F 12 68 7C 19 49 13 DF D3 8A 7D 65";
char url[] = "http://35.237.110.54/test/";
uint16_t timoutMillis = 500;
char strBuf[2048];

void lg(const char *s) {
    Serial.print(s);
}

void log(const char *s = "") {
    Serial.println(s);
}

bool httpTest() {
    std::unique_ptr <BearSSL::WiFiClientSecure> clientSSL = NULL;
    std::unique_ptr <WiFiClient> client = NULL;
    HTTPClient http;
    bool httpStatus;

    if (SSLFlag) {
        clientSSL.reset(new BearSSL::WiFiClientSecure);

        clientSSL->setBufferSizes(sizeof(strBuf), sizeof(strBuf));
        clientSSL->setFingerprint(SSLFingerprint);
        clientSSL->allowSelfSignedCerts();

        httpStatus = http.begin(*clientSSL, urlSSL);
    } else {
        client.reset(new WiFiClient);
        httpStatus = http.begin(*client, url);
[... 597 lines cut ...]
0x4021ebd4: sha2big_round at src/hash/sha2big.c line 124
0x4021eb7c: sha2big_round at src/hash/sha2big.c line 124
0x4021e68a: br_sha1_round at src/hash/sha1.c line 81
0x40204be4: BearSSL::WiFiClientSecure::available() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 378
0x40204df3: BearSSL::WiFiClientSecure::_connectSSL(char const*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 897
0x40207d38: String::charAt(unsigned int) const at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 550
0x402014d7: delay at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 46
0x40203a24: std::_Sp_counted_deleter  ::_Deleter   >, std::allocator , (__gnu_cxx::_Lock_policy)0>::_M_destroy() at /Users/atsarev/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/shared_ptr_base.h line 350
0x40204eba: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 223
0x40208728: WiFiClient::operator bool() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 349
0x40208384: init_done() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 181
0x40205ed4: HTTPClient::addHeader(String const&, String const&, bool, bool) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 959
0x402066bc: HTTPClient::writeToStreamDataBlock(Stream*, int) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1271
0x40206ba8: HTTPClient::~HTTPClient() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 126
0x40206ec1: HTTPClient::beginInternal(String, char const*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 320
0x40204057: std::__shared_ptr ::operator=(std::__shared_ptr &&) at /Users/atsarev/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/move.h line 177
0x4020676a: HTTPClient::writeToStreamDataBlock(Stream*, int) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 1251
0x4020296c: httpTest() at /Users/atsarev/dev/SmartBox/src/HTTPClient/HTTPClient.ino line 28
0x40206e00: HTTPClient::beginInternal(String, char const*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 300
0x40202d44: ESP8266WiFiSTAClass::begin() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 217
0x40206f10: HTTPClient::beginInternal(String, char const*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 320
0x40202d44: ESP8266WiFiSTAClass::begin() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 217
0x40202a61: wifi_dns_found_callback(char const*, ip4_addr const*, void*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp line 576
0x40207de4: String::lastIndexOf(String const&, unsigned int) const at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 638
```

AlexeySofree, 2019-02-20:
I made a simpler example, which works in 2.4.2
```cpp
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char *host = "35.237.110.54";
const char *url = "/test/";
const int httpsPort = 443;
const char *fingerprint = "26 24 FA 2B 26 1D DD 1F 2F 12 68 7C 19 49 13 DF D3 8A 7D 65";

bool httpTest() {
    WiFiClientSecure client;
    client.allowSelfSignedCerts();

    if (client.connect(host, httpsPort)) {
        Serial.println("connection OK");
    } else {
        Serial.println("connection failed");
        return false;
    }

    if (client.verify(fingerprint, host)) {
        Serial.println("certificate matches");
    } else {
        Serial.println("certificate doesn't match");
    }

    Serial.print("requesting URL: ");
    Serial.println(url);

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "User-Agent: BuildFailureDetectorESP8266\r\n" +
                 "Connection: close\r\n\r\n");

    while (client.connected()) {
        String line = client.readStringUntil('\n');
        if (line == "\r") {
            Serial.println("headers received");
            break;
        }
    }

    String line = client.readStringUntil('\n');

    Serial.println("reply:");
    Serial.println(line);
    Serial.println("");
}

void setup() {
    Serial.begin(115200);
    Serial.setDebugOutput(true);

    while (!Serial)
        yield();

    WiFi.begin();
}

void loop() {
    if (WiFi.status() == WL_CONNECTED) {
        httpTest();
    } else {
        Serial.println("Not connected");
    }
}
```

It works perfectly when I turn off my access point.
[... 640 lines cut ...]
0x40100518: _umm_free at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1300
0x40201945: _malloc_r at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/heap.c line 20
0x402061f8: loop_task(ETSEvent*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 131
0x40206319: __yield() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 97
0x40202c9b: WiFiClient::peekBytes(unsigned char*, unsigned int) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 302
0x4020474e: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 447
0x4021cc80: sha2big_round at src/hash/sha2big.c line 127
0x4021cc28: sha2big_round at src/hash/sha2big.c line 127
0x4021c736: sha2big_round at src/hash/sha2big.c line 124
0x40204980: BearSSL::WiFiClientSecure::_wait_for_handshake() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 524
0x40204b8f: BearSSL::WiFiClientSecure::_connectSSL(char const*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 936
0x402061f8: loop_task(ETSEvent*) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 131
0x402014d7: delay at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 46
0x40203874: std::_Sp_counted_deleter  ::_Deleter   >, std::allocator , (__gnu_cxx::_Lock_policy)0>::_M_destroy() at /Users/atsarev/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/ext/new_allocator.h line 110
0x40201945: _malloc_r at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/heap.c line 20
0x40204556: BearSSL::WiFiClientSecure::_installClientX509Validator() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 868
0x40202778: httpTest() at /Users/atsarev/dev/SmartBox/src/HTTPClient_old/HTTPClient_old.ino line 46
0x40100a4e: timer1_isr_handler at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_timer.c line 35
0x4020599c: HardwareSerial::available() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/HardwareSerial.cpp line 98
0x40202b94: WiFiClient::connect(char const*, unsigned short) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 126
0x40205c80: Print::println(String const&) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Print.cpp line 184
0x40202b94: WiFiClient::connect(char const*, unsigned short) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/WiFiClient.cpp line 126
0x402028b1: ESP8266WiFiGenericClass::mode(WiFiMode) at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp line 381
0x402062a4: init_done() at /Users/atsarev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 181
```

thjean, 2019-05-02:
I experienced the same (or at least a very similar) issue with SW WDT resets when the WiFi connection is lost during the client.connect() call. Connecting to an url with a long response time generated this issue on a regular base. After quite a lot of testing, I was able to narrow down the root cause of my crashes to the WiFi loss during a WiFiClientSecure connection. I did not double-check if a WiFiClient connection is affected as well.
I was finally able to reproduce the issue with this simple sketch:

```
#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID "SSID"
#define STAPSK  "PSK"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

void setup() {
  Serial.begin(115200);

  Serial.println();
  Serial.printf("Version info: %s\r\n", ESP.getFullVersion().c_str());
 
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  delay(2500);
  do_connect("xxxxx.xxx", 443);      // url that takes some time to connect
}

void do_connect(const char* host, int port) {
  uint32_t start, stop;
  
  Serial.print("\r\nconnecting to ");
  Serial.print(host);
  Serial.print(':');
  Serial.println(port);

  WiFiClientSecure client;
  client.setInsecure();
  client.setBufferSizes(1024, 512);

  start = millis();
  if (!client.connect(host, port)) {
    stop = millis();
    Serial.printf("connection failed, timeout: %d ms\r\n", stop - start);
  }

  if (client.connected()) {
    stop = millis();
    Serial.printf("Connection done in %d ms, disconnecting\r\n", stop - start);
  }
  client.stop();
}
``` 
With core revision ``` SDK:2.2.1(cfd48f3)/Core:unix-2.6.0-dev=-194000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-5-g25d5e81/BearSSL:6778687 ``` the sketch/debug output just before the crash is 
``` connecting to xxxx.xx:443
[hostByName] request IP for: xxxx.xx
[... 152 lines cut ...]
connection failed, timeout: 15796 ms
:ur 1
:del
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with SSID, channel 11
dhcp client start...
wifi evt: 0
ip:192.168.73.38,mask:255.255.255.0,gw:192.168.73.1
wifi evt: 3

connecting to xxxx.xx:443
[hostByName] request IP for: xxxx.xx
[hostByName] Host: xxxx.xx IP: xx.xx.xx.xx
:ref 1
BSSL:_connectSSL: start connection
```

mnunespt, 2019-05-14:
Hi,
I've a similar problem with versions 2.5.0 and 2.5.1
If i keep connection to WLAN alive, but kill gateway or connection to internet, the WifiClienteSecure hangs without error trigger.
I wish that a error triggers when internet connection lost. 
Can anyone help please? Many thanks.

earlephilhower, 2019-05-15:
Please try the latest GIT heat.  #6065 included a fix that enables a fst SSL disconnect response and, I think, will error out quickly and not hang.  Please report back so we can track.

thjean, 2019-05-16:
The fix seems to be effective for the case "WiFi disconnects while SSL-connect is ongoing", I do no longer observe SW WDT resets.

Using version ``` SDK:2.2.1(cfd48f3)/Core:unix-2.6.0-dev=-194000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-5-g25d5e81/BearSSL:6778687 ``` (begin of April), I can reproducibly see SW WDT resets when WiFi disconnects during the ongoing SSL connect with the sketch attached below. 
With the current git head ``` SDK:2.2.1(cfd48f3)/Core:2.5.1-10-g2dff28ab=20501010/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020 ```, the connect times out as expected. With debug set to SSL, the debug output is 
``` 
BSSL:_connectSSL: start connection
state: 5 -> 0 (0)
rm 0
BSSL:_wait_for_handshake: failed
BSSL:Couldn't connect. Error = 'Unknown error code.' 
```
 
The git head revision does no longer show the SW WDT resets since around begin of May. 

I used this sketch for reproducing the issue and testing the fix. A ticker disconnects the WiFi while the SSL-connect is ongoing. 
```
#include <ESP8266WiFi.h>
#include "Ticker.h"

#ifndef STASSID
#define STASSID "yourSSID"
#define STAPSK  "yourPSK"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

Ticker gWifiDisconnectTimer;

void setup() {
  Serial.begin(115200);

  Serial.println();
  Serial.printf("Version info: %s\r\n", ESP.getFullVersion().c_str());
 
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
}

void disconnectWiFi () {
  WiFi.disconnect();
  WiFi.begin(ssid, password);
}

void loop() {
  delay(5000);
  do_connect("api.github.com", 443);
}

void do_connect(const char* host, int port) {
  uint32_t start, stop;
  
  Serial.print("\r\nconnecting to ");
  Serial.print(host);
  Serial.print(':');
  Serial.println(port);

  WiFiClientSecure client;
  client.setInsecure();
  client.setBufferSizes(1024, 1024);

  gWifiDisconnectTimer.once_ms(500, disconnectWiFi);

  start = millis();
  if (!client.connect(host, port)) {
    stop = millis();
    Serial.printf("connection failed, timeout: %d ms\r\n", stop - start);
  }

  if (client.connected()) {
    stop = millis();
    Serial.printf("Connection done in %d ms, disconnecting\r\n", stop - start);
  }
  client.stop();

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
}

```

sislakd, 2019-05-16:
I found an interesting bug causing hang of WiFiClient and WiFiClientSecure which could be related - see #6104.

mnunespt, 2019-05-16:
> 
> 
> Please try the latest GIT head. #6065 included a fix that enables a fst SSL disconnect response and, I think, will error out quickly and not hang. Please report back so we can track.

Many many thanks, that solution works like a charm. I've been fighting this for weeks. Have made a workaround, but this solved my problem. I believe it can help many more people.
Thanks once again.

earlephilhower, 2019-05-16:
All credit goes to @sislakd for this specific fix.  I'll close this bug for now since it seems to have multiple confirmations.

closed 2019-05-16 by earlephilhower