esp8266/Arduino #5788: 2.5.0 SPIFFS fatal crash with Illegal Instruction on call to millis() within interrupt handler

ks12, 2019-02-19:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					ESP 12E on Nodemcu 1.0
- Core Version:			2.5.0
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			qio
- Flash Size: 		4MB/1MB]
- lwip Variant:   v2 Higher Bandwidth (no features)
- Reset Method:		--
- Flash Frequency:	--
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   --

### Problem Description

In 2.5.0 core, I am facing recurrent ESP crashes at the millis() line of the code:

void doCnt()                                                             // This is an interrupt handler function
{
    if ((millis() - CntLMillis) > cntBounce)                          // << This line of code
    {
        CntLMillis = millis();
        checkCnt = true;
    }
}

The crash decode shows that its a Fatal Exception 0 - Illegal Command. 
millis() is the only command on that line.
The stack shows a lot of SPIFFS activity then a esp_yield  and then this crash.

I think it has something to do with interrupt handling along with SPIFFS activity. I cannot stop my interrupt captures during SPIFFS writes, due to application requirements.

I upgraded from the 2.4.2 core to 2.5.0.
My same sketch was very stable in the older core.

After I shifted to the older code, things are stable again.

devyte, 2019-02-19:
The issue template requires a MCVE sketch to reproduce the issue. Without that, it is unlikely this will be looked at.
Please provide a minimal sketch.

faparo, 2019-03-01:
Hi devyte, i have the same problem. From one day to the next my sketch stopped working.
The description of the cause is the same as described by ks12:

"Fatal Exception 0 - Illegal Command.
millis() is the only command on that line.
The stack shows a lot of SPIFFS activity then a esp_yield and then this crash.".

If, within my sketch, I comment SPIFFS.begin () the program does not crash, the FS also stops working too.
I´m using Platformio/arduino IDE.
I´m new in this, veryyyy new but if you explain to me that it is an "MCVE sketch",
 I promise to help to the best of my ability.

faparo, 2019-03-01:
Hi devyte, it´s me again.
i found it "MCVE sketch: Minimial Complete Verifiable Example."
i will try to provide you the MCVE.

laercionit, 2019-03-21:
I have the same problem.

earlephilhower, 2019-07-24:
The bug has been oven for >6 months and no MCVE or even a decoded stack trace are available.  There's nothing we can look at here.  Closing.

closed 2019-07-24 by earlephilhower