esp8266/Arduino #5792: Commit esp8266/Arduino@9790e1c python issue

reaper7, 2019-02-20:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 10]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[default]
- Flash Size: 		[4MB/0MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
1
After commit [9790e1c](https://github.com/esp8266/Arduino/commit/9790e1cb7cfa01ac74056fb835e6282af123aba0) , archive **python-3.7.2.post1-embed-win32.zip** is unpacked into wrong directory.
Should be unpacked to _esp8266/tools/python/_
but is unpacked to parent dir _esp8266/tools/_

Compilation break with error -> debug messages below "_1. Python is missing_"

2
After manually moved all python related files to _esp8266/tools/python/_ ,
sketch compiles without problems but attempt to load fails with an error -> debug massages below "_2. Upload problem_"

related:
https://github.com/esp8266/Arduino/commit/9790e1cb7cfa01ac74056fb835e6282af123aba0#commitcomment-32366937
https://github.com/esp8266/Arduino/issues/5789 FIXED with https://github.com/esp8266/Arduino/pull/5791

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

examples->digital->BlinkWithoutDelay.ino

### Debug Messages

1. Python is missing:
```
C:\Programy\arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Programy\arduino\hardware -hardware C:\Users\reaper7\AppData\Local\Arduino15\packages -tools C:\Programy\arduino\tools-builder -tools C:\Programy\arduino\hardware\tools\avr -tools C:\Users\reaper7\AppData\Local\Arduino15\packages -built-in-libraries C:\Programy\arduino\libraries -libraries C:\Programy\arduino_projekty\libraries -fqbn=esp8266com:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10808 -build-path C:\TEMP\arduino_build_697025 -warnings=none -build-cache C:\TEMP\arduino_cache_809626 -prefs=build.warn_data_percentage=75 -verbose C:\Programy\arduino\examples\02.Digital\BlinkWithoutDelay\BlinkWithoutDelay.ino
C:\Programy\arduino\arduino-builder -compile -logger=machine -hardware C:\Programy\arduino\hardware -hardware C:\Users\reaper7\AppData\Local\Arduino15\packages -tools C:\Programy\arduino\tools-builder -tools C:\Programy\arduino\hardware\tools\avr -tools C:\Users\reaper7\AppData\Local\Arduino15\packages -built-in-libraries C:\Programy\arduino\libraries -libraries C:\Programy\arduino_projekty\libraries -fqbn=esp8266com:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10808 -build-path C:\TEMP\arduino_build_697025 -warnings=none -build-cache C:\TEMP\arduino_cache_809626 -prefs=build.warn_data_percentage=75 -verbose C:\Programy\arduino\examples\02.Digital\BlinkWithoutDelay\BlinkWithoutDelay.ino
Using board 'nodemcuv2' from platform in folder: C:\Programy\arduino\hardware\esp8266com\esp8266
Using core 'esp8266' from platform in folder: C:\Programy\arduino\hardware\esp8266com\esp8266
Detecting libraries used...
"C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/lwip2/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\TEMP\\arduino_build_697025/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10808 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266\\cores\\esp8266" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266\\variants\\nodemcu" "C:\\TEMP\\arduino_build_697025\\sketch\\BlinkWithoutDelay.ino.cpp" -o nul
Generating function prototypes...
"C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/lwip2/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\TEMP\\arduino_build_697025/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10808 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266\\cores\\esp8266" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266\\variants\\nodemcu" "C:\\TEMP\\arduino_build_697025\\sketch\\BlinkWithoutDelay.ino.cpp" -o "C:\\TEMP\\arduino_build_697025\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Programy\\arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\TEMP\\arduino_build_697025\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Kompilowanie szkicu...
"C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/lwip2/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\TEMP\\arduino_build_697025/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10808 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266\\cores\\esp8266" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266\\variants\\nodemcu" "C:\\TEMP\\arduino_build_697025\\sketch\\BlinkWithoutDelay.ino.cpp" -o "C:\\TEMP\\arduino_build_697025\\sketch\\BlinkWithoutDelay.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/python/python.exe" "C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/signing.py" --mode header --publickey "C:\\Programy\\arduino\\examples\\02.Digital\\BlinkWithoutDelay/public.key" --out "C:\\TEMP\\arduino_build_697025/core/Updater_Signing.h"
exec: "C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/python/python.exe": file does not exist
Blad kompilacji dla plytki NodeMCU 1.0 (ESP-12E Module).
```

[... 9 lines cut ...]
"C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/lwip2/include" "-IC:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\TEMP\\arduino_build_697025/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10808 -DARDUINO_ESP8266_NODEMCU [... cut ...]
"C:\\Programy\\arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-size" -A "C:\\TEMP\\arduino_build_697025/BlinkWithoutDelay.ino.elf"
Szkic uzywa 257112 bajtów (24%) pamieci programu. Maksimum to 1044464 bajtów.
Zmienne globalne uzywaja 26548 bajtów (32%) pamieci dynamicznej, pozostawiajac 55372 bajtów dla zmiennych lokalnych. Maksimum to 81920 bajtów.
Traceback (most recent call last):
  File "C:\Programy\arduino\hardware\esp8266com\esp8266/tools/upload.py", line 24, in <module>
    pid  = os.fork()
AttributeError: module 'os' has no attribute 'fork'
AttributeError: module 'os' has no attribute 'fork'
```

closed 2019-02-20 by d-a-v

reaper7, 2019-02-20:
The archive is still incorrectly unpacked
Look at dirs struct after _python get.py_ command:
[image]

and what get.py rename file (instead dir):
[image]

so... I changed dir name from python-3.7.2.post1 to python
and restored _asyncio.pyd file to python dir.

Compilation OK but second problem still exist!
```
Traceback (most recent call last):
  File "C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/upload.py", line 24, in <module>
    pid  = os.fork()
AttributeError: module 'os' has no attribute 'fork'
AttributeError: module 'os' has no attribute 'fork'
```
os.fork works on windows10 ???

btw, why this issue is closed without any user feedback?

earlephilhower, 2019-02-20:
@reaper7, merges automatically close issues here on GH.  It's not like Bugzilla or some other tools I've used that require the original submitter to check and sign off.

Anyway, as others have pointed out, I re-uploaded the original packaged ZIP and not the one w/the proper subdir structure for Arduino in my rush out the door this AM.  D'oh.

I have fixed that and verified with my own eyeballs just now with #5797 that python does really expand into tools/python/.

Could you please delete your whole tools dir, "git update" to reset everything to fresh checkout state, apply the PR, and give it a go?  If not, let's get comments tracked in that PR so we can clean it up.

The original esptool.py may have gotten bitten by bit (or brain) rot in the month or so since I stopped looking at it, and all active maintainers are Linux/Mac first (I need to spin up a Win VM w/USB passthru for testing), so Windows' quirks are something of a pain...

Thx
-EFP3

reaper7, 2019-02-20:
so... #5797 solves problem with python dirs.
as @d-a-v wrote, compilation and signing OK
but still cant upload:
```
...
Linking everything together...
"C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\\tmp\\arduino_build_225980/local.eagle.app.v6.common.ld"
"C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -fno-exceptions -Wl,-Map "-Wl,C:\\tmp\\arduino_build_225980/Blink.ino.map" -g -Wall -Wextra -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/lib" "-LC:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/lib/NONOSDK221" "-LC:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/ld" "-LC:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" -Teagle.flash.4m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o "C:\\tmp\\arduino_build_225980/Blink.ino.elf" -Wl,--start-group "C:\\tmp\\arduino_build_225980\\sketch\\Blink.ino.cpp.o" "C:\\tmp\\arduino_build_225980\\core\\core.a" -lhal -lphy -lpp -lnet80211 -llwip2-536-feat -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group "-LC:\\tmp\\arduino_build_225980"
"C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/python/python.exe" "C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/elf2bin.py" --eboot "C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/bootloaders/eboot/eboot.elf" --app "C:\\tmp\\arduino_build_225980/Blink.ino.elf" --flash_mode dio --flash_freq 40 --flash_size 4M --path "C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin" --out "C:\\tmp\\arduino_build_225980/Blink.ino.bin"
Creating BIN file "C:\tmp\arduino_build_225980/Blink.ino.bin" using "C:\tmp\arduino_build_225980/Blink.ino.elf"

"C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/python/python.exe" "C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/signing.py" --mode sign --privatekey "C:\\PROGRAMY\\Arduino\\examples\\01.Basics\\Blink/private.key" --bin "C:\\tmp\\arduino_build_225980/Blink.ino.bin" --out "C:\\tmp\\arduino_build_225980/Blink.ino.bin.signed"
"C:\\PROGRAMY\\Arduino\\hardware\\esp8266com\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-size" -A "C:\\tmp\\arduino_build_225980/Blink.ino.elf"
Szkic używa 257256 bajtów (24%) pamięci programu. Maksimum to 1044464 bajtów.
Zmienne globalne używają 26556 bajtów (32%) pamięci dynamicznej, pozostawiając 55364 bajtów dla zmiennych lokalnych. Maksimum to 81920 bajtów.
Traceback (most recent call last):
  File "C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/upload.py", line 24, in <module>
    pid  = os.fork()
AttributeError: module 'os' has no attribute 'fork'
AttributeError: module 'os' has no attribute 'fork'
```

earlephilhower, 2019-02-20:
https://docs.python.org/3/library/os.html#os.fork

Windows is a right pain.

Python subprocess or multiprocessing should be able to make this work, but I'll need to think about it since I need to push in the pyserial path addition to the code stream.

reaper7, 2019-02-20:
@earlephilhower - what is your suggestion, I should temporarily switch to a stable version 2.5.0b3 from board manager or wait for a solution for current git version?

earlephilhower, 2019-02-20:
PRs always welcome. :)  It's gonna be several hours before I myself will have time to try and fake os.fork() in the uploader script.  2.5.0-rel doesn't have the esptool.py PR, so you can use that.

PR #5796 (2019-02-20): Use local copy of python distro to include a subdir
fix #5792