esp8266/Arduino #5793: Can't get MDNS.installServiceQuery to work

bikemike, 2019-02-20:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.5.0]
- Development Env:		Platformio
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			d1_mini
- Flash Mode:			[not sure]
- Flash Size: 		[4MB]
- lwip Variant:   [not sure]
- Reset Method:		[not sure]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

The old queryService function works as expected but the new installServiceQuery does not.

In the test code below the answerCount is always 0 unless I uncomment the queryService call.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

#include <ESP8266WiFi.h> 
#include <ESP8266mDNS.h>

const char* ssid = "xxxxx"; //your WiFi Name
const char* password = "xxxxx";  //Your Wifi Password

MDNSResponder::hMDNSServiceQuery hServiceQuery  = nullptr;

void serviceQueryCallback(const MDNSResponder::MDNSServiceInfo& mdnsServiceInfo, MDNSResponder::AnswerType answerType, bool p_bSetContent)
{
  Serial.println("got a callback");
}

void setup()
{
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  
  if (!MDNS.begin("ESPChromecastHelper"))
  {
    Serial.println("Error setting up mDNS");
  }
  
  hServiceQuery = MDNS.installServiceQuery("googlecast", "tcp", &serviceQueryCallback);
  if (!hServiceQuery)
    Serial.println("Error installing service");

}

uint32_t nextQueryTime   = 0;

void checkMDNS(String deviceName)
{
  uint32_t timeNow = millis();
  if (nextQueryTime < timeNow)
  {
    int n = MDNS.answerCount(hServiceQuery);

    Serial.print("Number of MDNS answers:");
    Serial.println(n);
    
  /*   if (0 == n)
    {
      n = MDNS.queryService("googlecast", "tcp");
      Serial.print("Number of MDNS answers:");
      Serial.println(n);
    }
 */    
    
    nextQueryTime = timeNow + 5000; // 5 seconds                     
  }
}

void loop()
{
    checkMDNS("Device Name");

    MDNS.update();
}

```

### Debug Messages

```
Number of MDNS answers:0
```

LaborEtArs, 2019-02-20:
Please enable '#define DEBUG_ESP_MDNS_INFO' in 'LEAmDNS_Priv.h' (around line 50).
In general it is recommended to install service queries, ... only after probing is finished (eg. in the probeResult callback), see the LEAmDNS examples.

bikemike, 2019-02-20:
Thanks. I will give that a try tonight. The documentation I followed for setting up a dynamic query is in LEAmDNS.h and it is incomplete I guess:

https://github.com/esp8266/Arduino/blob/82789d201c7200618fdee1c8bd02baf631ed5e12/libraries/ESP8266mDNS/src/LEAmDNS.h#L57

It doesn't say when/where to install service queries and it doesn't specify whether MDNS.update() should be called in the loop (only the "For presenting services:" section does). Also, the signature for the serviceQueryCallback in that comment is not correct.

devyte, 2019-02-21:
@bikemike the MDNS.update() call in the loop is a must, and has always been.
About installServiceQuery(), my own experiments showed exactly what @LaborEtArs said: the only sane place to call it is in the probe callback. See the serviceMonitor example.
Closing.

closed 2019-02-21 by devyte

bikemike, 2019-02-21:
I modified my code and it still wasn't working. Then, by chance, I held my finger over the wifi antenna and everything started to work. It turns out the problem was with my device and nothing to do with this new library. I even reverted my code to what is above (installing the service query in setup()) and it works.

Thanks for the help and I'm sorry to have wasted your time. This new mDNS library is very nice!