esp8266/Arduino #5799: DNS_MAX_NAME_LENGTH changed from lwip 1.4 to 2.0

reglos, 2019-02-21:
DNS_MAX_NAME_LENGTH for lwip v1.4 "Higher Bandwith" is defined as 256.
In lwip v2.0 it is set to 128 for all variants. 

As RFC1035 allows for dns hostnames up to 255 bytes, there should be an option to address these. 
I suggest setting it to 256 for all the "higher bandwith" lwip variants and leave it at 128 for the "lower memory" options.

Why not stay with 1.4 "Higher Bandwith"? There is a bug in lwip 1.4. After several hundered dns requests it suddenly starts to send out random garbage udp packets to the dns-server as I can see in the logfiles of my nameserver and verified by wireshark.

d-a-v, 2019-02-21:
This is an harcoded parameter in lwIP, that was lowered because it takes this ram multiplied by the number of entries in the dns client cache.
Do you need fqdn with more than 128 bytes ?

reglos, 2019-02-21:
Yes, I have to lookup addresses in a subnet which can use the full possible namelength permitted by RFC1035. Took me quite a while to find out why my sketch worked on all test-cases (short names) but when deployed failed on some occasions, because I didn't expect non-conformant behaviour of a basic network functionality.  

Can lwip v2 somehow be compiled with a 
#define DNS_MAX_NAME_LENGTH 256 ?
and a very small cache size?

d-a-v, 2019-02-21:
Maybe it's time for https://github.com/esp8266/Arduino/issues/3740#issuecomment-338863188

In the meantime you can locally recompile lwip2 with this updated limitation.
```
cd tools/sdk/lwip2
make download
cd builder/glue-lwip/arduino
vi lwipopts.h      (DNS_MAX_NAME_LENGTH line 1108)
cd ../../..
make install
```

devyte, 2019-02-21:
@d-a-v I agree about the comment in #3740.
@reglos you are the first user ever since lwip2 was integrated to encounter this. The precompiled lwip used by the core won't be changed for full rfc conformance, because it reduces available heap for every other user, and they don't need it. Also, you do have the option to build your own custom lwip, as @d-a-v explained.
However, we'll pursue the solution referenced above with lwip upstream and bring it in later on. That would be the best of both worlds, I think.

reglos, 2019-02-21:
Thanks @d-a-v, compiling lwip2 with modified defines worked fine for me. I had been searching quite a while for the recipe...  I had to install the git version first, correct? In the boards manager version I did not find the lwip2 sources. Are there any problems to expect, having both versions now installed in the Arduino IDE?  

@ devyte The application is indeed  rather exotic, but there are applications where you want to put a lot of information into a hostname. Until the dynamic allocation makes it into lwip2, I am fine with compiling myself.

d-a-v, 2020-11-09:
> I am fine with compiling myself.

:+1:

closed 2020-11-09 by d-a-v