esp8266/Arduino #5801: Cannot override default ld script after version 2.5.0

joaoTrevizoli, 2019-02-21:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			2.5.0
- Development Env:		Platformio
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			n/a
- Flash Size: 		4MB
- lwip Variant:  v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   115200

### Problem Description

If i change the flash size by overrinding the ld script with any of the available ones the board keeps wdt reseting (cause:4, boot mode:(1,6)), this problem never happened in previous versions, just now after I updated core to version 2.5.0. 

I am using in platformio.ini the following settings:

[env:nodemcuv2_dev]
platform = espressif8266
framework = arduino
board = nodemcuv2
upload_speed = 115200
monitor_speed = 115200
build_flags =
  -Wl,-Teagle.flash.2m.ld
lib_deps =
  31
  19
  64
  83
  576
  663
  748

I tried a few ld scripts available at:
https://github.com/esp8266/Arduino/tree/master/tools/sdk/ld

none seems to work

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() 
{
  Serial.begin(115200);
}

void loop() 
{
  Serial.println("Hello world!")
}

```

devyte, 2019-02-23:
I don't understand your use case. Why are you trying to override ld scripts with your own?

joaoTrevizoli, 2019-02-25:
I am actually trying to use one of the flashmodes availables at: https://github.com/esp8266/Arduino/tree/master/tools/sdk/ld . The reason I was using and actually am using it (I downgraded my core version to 1.4.2) is because I use the flash as a data backup for when Inteternet is Down, which is a common problem here. With these ld scripts I can free more space,  up to 4mb. After the update to core 2.5 the esp8266 keeps reseting.

@devyte you can see an example of usage in this link: https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

devyte, 2019-02-25:
The ld scripts are currently generated by a tool, and I think it's not enough to just change which one you use. I think there are defines as well (I'm not positive of this). 
I suggest looking at how the IDE build commands are put together when switching flash size in the IDE menu, then figuring out the equivalent for platformio.
Sorry, I can't help you further.
Closing due to not a core issue.

closed 2019-02-25 by devyte