esp8266/Arduino #5804: How to test esptool 2.6 with Arduino

scottchiefbaker, 2019-02-21:
I see `esptool` version 2.6.0 has been in the tree since [October](https://github.com/espressif/esptool/commit/7460b2d699784e5debf39228ff27ae89cff68f3c#diff-2e7375ee839ab724de2eec9abc81b312). This version offers a lot of improvements over previous `esptool` versions. Most namely that it supports compression of images on the fly. This greatly increases upload speed, and would really help my workflow.

How do I test using the new version to upload sketches? I've only ever installed boards via the UI in the Arduino IDE. Is it as simple as just cloning this repo to `~/.arduino15/packages/esp8266`?

In other words, how do I help you test the next release that's **not** available yet via the Arduino IDE?

scottchiefbaker, 2019-02-21:
If I install the official release from the IDE I get:

    ~/.arduino15/packages/esp8266/hardware
    ~/.arduino15/packages/esp8266/tools

I'm assuming if I clone this repo it should go in to `hardware` and then I download the tools via the pythong script to `tools`?

oddstr13, 2019-02-23:
You should clone this repo into your arduino sketch folder, under hardware - not where the boards manager installs the packages.

devyte, 2019-02-23:
No, you're not on the right track. The instructions are correct, the ~/.arduino15 dir is different. When you install git, the correct place is ~/arduino, or whatever dir you specify in the IDE preferences (I personally have ~/workspace/arduino).

My own setup is as follows (in case anyone reading this is interested in an alternate setup):
* in the above dir, I have the dir structure shown in the instructions: ~/workspace/arduino/hardware/esp8266com
* I forked the repo from the webpage to have my own copy
* I cloned this main repo into ~/workspace/Arduino.esp8266
* I cloned my forked repo into ~/workspace/Arduino.devyte
* I created a symlink inside esp8266com called esp8266 that points to one of the two above.
For official business and testing:
```
cd ~/workspace/arduino/hardware/esp8266com
symlink -s ../../../Arduino.esp8266 esp8266
```
For messing around with my own fork, or making PRs:
```
cd ~/workspace/arduino/hardware/esp8266com
rm esp8266
symlink -s ../../../Arduino.devyte esp8266
```

In other words, I can switch between the two.
That's advanced (and paranoid) usage, though, it's probably enough for you to just clone directly into esp8266com/esp8266, as explained in the git instructions (basic).

Closing as not an issue.

closed 2019-02-23 by devyte

scottchiefbaker, 2019-02-23:
Ah OK this makes a lot more sense. I have the repo working and recognized by Arduino now. When I attempt to upload to my 8266 the `upload.py` I get an error:

    Error in command line, need pyserial path as 1st arg and esptool path as 2nd.

When I check the `/esptool` and `/pyserial` directories they're totally empty? Even after running `get.py`.

I think there is room for some cleanup on the documentation. I'll be happy to whip up a PR after I get this working 100%.

scottchiefbaker, 2019-02-24:
Oh I see `/esptool` and `/pyserial` are sub-modules. I just submitted a pull-request (#5816) to clarify this.