esp8266/Arduino #5806: Weird problems with tcp

cziter15, 2019-02-22:
Hello,

I'm fighting an issue in my mqtt client. I'm using PubSubClient library. It all runs on esp12f module with 2.5.0 release.

The behaviour is strange, my mqtt connection timeouts and it depends on lwip version.

I've tried lwip2 higher bandwidth/lower memory and it does not matter, connection timeouts few times per hour.

When switched to lwip 1.4 higher bandwidth, it timeouts only once a day, but still.

WiFi connection is not being lost. Router seems to be connected without interruptions.

Tried earlier releases, same results.

devyte, 2019-02-23:
Try latest git. Issues were found in SDK pre3 which is included in 2.5.0, so the SDK has been reverted to 2.2.1 after 2.5.0 release. Older core releases have other issues that have been fixed since, so reverting the core isn't enough either (you could be switching one reason that causes timeout for a different one that also causes timeout).

cziter15, 2019-02-23:
Latest git, boootloop with this log:

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vffffffff
~ld`

cziter15, 2019-02-23:
Lastest git fails at SPIFFS.

> Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
> PC: 0x40215f48
> EXCVADDR: 0x402497b1
> 
> Decoding stack results
> 0x4020b148: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\spiffs_hal.cpp line 45
> 0x4020b204: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\spiffs_hal.cpp line 102
> 0x4020b340: spiffs_hal_erase(unsigned int, unsigned int) at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\spiffs_hal.cpp line 168
> 0x4020ac60: SPIFFSImpl::open(char const*, fs::OpenMode, fs::AccessMode) at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\spiffs_api.cpp line 30
> 0x40209db8: fs::FS::open(char const*, char const*) at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\FS.cpp line 221
> 0x4020139d: Config::openConfigObject() at C:\Users\lenovo\AppData\Local\Temp\VMBuilds\OOKBridge\esp8266_espino\Debug\Config\Config.cpp line 12
> 0x4020115c: BridgeApp::begin() at C:\Users\lenovo\AppData\Local\Temp\VMBuilds\OOKBridge\esp8266_espino\Debug\BridgeApp.cpp line 16
> 0x40100124: millis() at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring.cpp line 186
> 0x4020926e: __yield() at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 100
> 0x402010a3: setup() at C:\Users\lenovo\Documents\Visual Studio 2015\Projects\OOKBridge\OOKBridge/OOKBridge.ino line 25
> 0x402010a9: setup() at C:\Users\lenovo\Documents\Visual Studio 2015\Projects\OOKBridge\OOKBridge/OOKBridge.ino line 26
> 0x40209234: loop_wrapper() at C:\Users\lenovo\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 122
>

devyte, 2019-02-23:
Your logs show 2.5.0. Either you're still using the release, or you installed git in a non-supported way.

cziter15, 2019-02-23:
I just checked latest git into 2.5.0 dir.

devyte, 2019-02-23:
That's not a supported installation method. Please read the docs.

cziter15, 2019-02-28:
Followed the docs, installed latest git step by step.
Same crash.

cziter15, 2019-03-12:
I've found that my router antenna was not original one.

Antenna has been connected using wrong SMA socket pair (female-to-female), which obviously didn't work. That was possible cause of all of my problems. Now I have stable ~ -30dBm signal.

My module is up for 41 hours now, without disconnection. However, I'd like to wait some more time, to definitely confirm that was caused by antenna.

cziter15, 2019-04-07:
Moved to another router firmware Openwrt, that allowed me to do some diag tests. There was a problem with LTE rsrp parameter, that sometimes cause timeouts, but doesn't restarts whole LTE session.

Changed network operator to another one. No problems, tcp connection is alive for 6 days now.

Conclusion: Never trust "conection time" status on router's configuration panel.

Please close this issue.

closed 2019-04-10 by devyte