esp8266/Arduino #5812: ESP8266 Connection problems

KGSoftware, 2019-02-23:
### Basic Infos

Hi
This is my first post here, I hope I do it in the right format.

Normaly I can connect with ESP8266 Wroom2 Module in STA mode to any access point.
But in one case it's not possible.

Below I posted the diagnostic output plus the code of the sketch.

Can anyone explain the meaning from the diagnostic output?
And what's the problem in this case?

Many thanks you in advance!!!

Best regards Klaus

#### Platform

- Hardware:			         ESP8266 Wroom 2
- Core Version:			2.2.0, and 2.5.0-betha3
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:  ?
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#ifndef STASSID
#define STASSID "skynet"
#define STAPSK  "************"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

ESP8266WebServer server(80);

const int led = 12;
int blinkCounter = 0;
int stateCounter = 0;
uint32_t realSize;
uint32_t ideSize;
FlashMode_t ideMode;

void handleRoot() {
  digitalWrite(led, HIGH);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, LOW);
}

void handleNotFound() {
  digitalWrite(led, HIGH);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, LOW);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);
  Serial.begin(9600);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.print("SSID: ");
  Serial.println(ssid);
  Serial.print("Passwort: ");
  Serial.println(password);

  realSize = ESP.getFlashChipRealSize();
  ideSize = ESP.getFlashChipSize();
  ideMode = ESP.getFlashChipMode();

  Serial.printf("Flash real id:   %08X\n", ESP.getFlashChipId());
  Serial.printf("Flash real size: %u bytes\n\n", realSize);

  Serial.printf("Flash ide  size: %u bytes\n", ideSize);

    
  Serial.printf("Flash ide speed: %u Hz\n", ESP.getFlashChipSpeed());

  
  Serial.printf("Flash ide mode:  %s\n", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));
  
  if (ideSize != realSize) {
    Serial.println("Flash Chip configuration wrong!\n");

  } else {
    Serial.println("Flash Chip configuration ok.\n");
  
  }

  

  Serial.setDebugOutput(true);

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");

    
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) 
{
  server.handleClient();
  MDNS.update();
  if(WiFi.status() == WL_CONNECTED)
  {
    stateCounter = stateCounter + 1;
    blinkCounter = blinkCounter + 1;
    if(blinkCounter <= 5000)
    {
      digitalWrite(led, LOW);
    }
    else
    {
      digitalWrite(led, HIGH);
      if(blinkCounter >= 10000)
      {
        blinkCounter = 0; 
      } 
    } 

    if(stateCounter >= 100000)
    {
      stateCounter = 0;
      WiFi.printDiag(Serial); 
    }
  }  
}
```

### Debug Messages

```

<<25>??<3>?<29>???<\f>??<30>?<\r><\n>
SSID: skynet<\r><\n>
Passwort: ************<\r><\n>
Flash real id:   001640A1<\n>
Flash real size: 4194304 bytes<\n>
<\n>
Flash ide  size: 4194304 bytes<\n>
Flash ide speed: 40000000 Hz<\n>
Flash ide mode:  DIO<\n>
Flash Chip configuration ok.<\n>
<\r><\n>
scandone<\n>
state: 0 -> 2 (b0)<\n>
state: 2 -> 3 (0)<\n>
state: 3 -> 0 (1)<\n>
wifi evt: 1<\n>
STA disconnect: 203<\n>
..reconnect<\n>
f 0, ....scandone<\n>
state: 0 -> 2 (b0)<\n>
state: 2 -> 3 (0)<\n>
state: 3 -> 0 (1)<\n>
wifi evt: 1<\n>
STA disconnect: 203<\n>
..reconnect<\n>
f -180, ....scandone<\n>
state: 0 -> 2 (b0)<\n>
..state: 2 -> 0 (2)<\n>
reconnect<\n>
f r0, wifi evt: 1<\n>
STA disconnect: 2<\n>
....scandone<\n>
state: 0 -> 2 (b0)<\n>
state: 2 -> 3 (0)<\n>
state: 3 -> 0 (1)<\n>
wifi evt: 1<\n>
STA disconnect: 203<\n>
..reconnect<\n>
f 0, .....scandone<\n>
state: 0 -> 2 (b0)<\n>
..state: 2 -> 0 (2)<\n>
reconnect<\n>
f -180, wifi evt: 1<\n>
STA disconnect: 2<\n>
....scandone<\n>
state: 0 -> 2 (b0)<\n>
state: 2 -> 3 (0)<\n>
state: 3 -> 0 (1)<\n>
wifi evt: 1<\n>
STA disconnect: 203<\n>

```

devyte, 2019-02-23:
#5784, closing.

closed 2019-02-23 by devyte

KGSoftware, 2019-02-24:
Hi devyte
Is the problem in this case the wpa2-enterprise issue?
Did you see this in the debug output?

Thank you in advance..

devyte, 2019-02-24:
No, this is most likely a duplicate of #5784, which is the master issue for the connectivity problems that came with sdk pre3.

KGSoftware, 2019-02-25:
Hi devyte
Thank you for your quick response.
Is there a good documentation for the wifi debug messages i posted in the first comment?
I searched hours, but without a meaningful result.

Again the debug message was like below:
> scandone<\n>
> state: 0 -> 2 (b0)<\n>
> state: 2 -> 3 (0)<\n>
> state: 3 -> 0 (1)<\n>
> wifi evt: 1<\n>
> STA disconnect: 203<\n>
> ..reconnect<\n>
> f 0, ....scandone<\n>
> state: 0 -> 2 (b0)<\n>

Because at my homenetwork the esp8266 could connect to my wifi network.
But in another case (this issue) it was not possible (it's a network from a company) and therfore i thougt its maybe a WPA2-Enterprise issue.

Can you give a guess?
Or should i only downgrade to arduino-core 2.4.0?

Thank you in advance

devyte, 2019-02-25:
The state messages come from the sdk. I don't think there is formal documentation about sdk debug messsges.
If you're trying to connect to wpa2-enterprise, there is only limited compatibility in the sdk, so a lot of cases won't work. That is the main reason why it's not formally supported in our core yet.
If both your home and work networks are wpa2, then having one work and not the other may be new information for the connectivity issue in #5784 .

chuancqc, 2020-05-02:
You must confirm whether your wireless network firewall blocks ESP devices