esp8266/Arduino #5818: OTA update not working over HTTPS

dmdt, 2019-02-24:
### Basic Infos

- [*] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [*] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [*] I have tested that the issue is present in current master branch (aka latest git).
- [*] I have searched the issue tracker for a similar issue.
- [*] If there is a stack dump, I have decoded it.
- [*] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12E
- Core Version:			2.5.0
- Development Env:		Arduino IDE/vscode w/ extension
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Module)
- Flash Mode:			didn't find, through CH340
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I trying update ESP by OTA from my server over HTTPS. Server is Apache/2.4.6 with PHP 7.3 and Let's Encrypt Authority X3 certificate. Server side script taken from documentation. Basically nothing happens at all (Debug Messages and Log from server). In the server logs package size rarely is 32768 instead 16384. Once I catched wdt reset - decoded stack below.

Strange fact - I get successfully updated ESP only when enabled debug on level "CORE+WIFI+HTTP_UPDATE+UPDATER+OTA+OOM", not from first time, it takes several attempts. But it is updates every time I try! On other debug levels it still not updating.

I tried update without server script, directly pointing .bin location (not working). Often I gets "Read Timeout" error and tried increase timeout for 10 sec, but it isn't helped.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <WiFiClientSecureBearSSL.h>

const char *fingerprint = "69:D6:81:6C:C9:B3:2F:78:C4:66:44:36:A5:79:49:99:DC:A9:DD:1F";

void setup() {

  Serial.begin(115200);
  WiFi.begin(APSSID, APPSK);
    Serial.print("\nConnecting to wifi..\n");
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }
    Serial.println("WiFi Connected!");
}

void loop() {
    std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);
    client->setFingerprint(fingerprint);
    ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);
    t_httpUpdate_return ret = ESPhttpUpdate.update(*client, "https://host/update.php", "1.1.1");
    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

[... 70 lines cut ...]
0x4021fbfe: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1305
0x40204aa0: BearSSL::WiFiClientSecure::_wait_for_handshake() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 526
0x40204c83: BearSSL::WiFiClientSecure::_connectSSL(char const*) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 935
0x40207e98: esp_yield() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 91
0x40201627: delay at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring.c line 51
0x402038c1: WiFiClient::connect(IPAddress const&, unsigned short) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/ClientContext.h line 136
0x40204d4a: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 223
0x40209984: HTTPClient::connected() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 448
0x402088b4: HTTPClient::connect() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 1055
0x402074fb: String::reserve(unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 146
0x40208e20: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 593
0x4020769d: String::operator=(char const*) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 243
0x40211988: operator new[](unsigned int) at ../../../../../dl/gcc-xtensa/libstdc++-v3/libsupc++/new_opv.cc line 33
0x40208767: HTTPClient::collectHeaders(char const**, unsigned int) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 981
0x40208ece: HTTPClient::GET() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 526
0x40205e51: ESP8266HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266httpUpdate\src\ESP8266httpUpdate.cpp line 290
0x40100d7c: free at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\umm_malloc\umm_malloc.c line 1760
0x4020746c: String::invalidate() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 141
0x40206039: ESP8266HTTPUpdate::update(WiFiClient&, String const&, String const&) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266httpUpdate\src\ESP8266httpUpdate.cpp line 94
0x40207400: UpdaterClass::printError(Print&) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\Updater.cpp line 512
0x4020757a: String::String(char const*) at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\WString.cpp line 36
0x4020287c: loop() at C:\Rep\Dev\Arduino\Abug/Abug.ino line 94
0x4020280c: setup() at C:\Rep\Dev\Arduino\Abug/Abug.ino line 88
0x40207f44: loop_wrapper() at C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 125
```

earlephilhower, 2019-05-01:
I'm unable to reproduce any problem with this on my own testing.  I've pointed the app (seems like a copy of the example?) directly to a private https server (and updated the fingerprint, obviously) and had it pull file.bin directly.  It works every time and then reboots into the new sketch.

Please past the full decode, including the header lines (i.e. where it actually crashed, whether there were OOM errors, etc.) on another run with GIT head.  The current issue didn't list where it actually crashed.

Anything to the pattern (like, it tries and fails to find a new file 3 times but then on the 4rd itry it gets it and crashes or something)?

dmdt, 2019-06-04:
I'm reverted to 2.4.2 to fix this problem. Now I use 2.5.2, where updates (seems) works fine.
But there is some problem from 2.5.0. When I send GET requests periodically - responses doesn't update. I.e. I get response "1", then I change data on server ("1" -> "2"), but responses still go with "1" payload. When I reboot board responses came with new data "2", but when I change again - responses still "2".
And when HTTP reuse set to false odd responses come with 200 code, even responses with -2 code (Can't send header), and both come without payload.
And also I'd like to note that on 2.4.2 requests works properly, so problem caused not by server.
'''
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

#define SSID 
#define PASS 

HTTPClient https;
BearSSL::WiFiClientSecure client;

void setup() {
    Serial.begin(115200);
    WiFi.begin(SSID,PASS);
    while(WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }
    Serial.println("\nWifi connected!");

    uint8_t cert[20] = {184,120,14,91,203,255,205,98,26,130,214,144,95,34,236,197,210,208,31,245};
    client.setFingerprint(cert);
    https.setReuse(true);
    String url = "https://apilink";
    https.begin(client,url);
    Serial.println(https.GET());
    Serial.println(https.connected());
    for(uint8_t i = 0; i < 10; i++) {
        Serial.print(String(i)+" : ");
        Serial.println(https.GET());
        Serial.println(https.getString());
        client.flush();
        delay(1000);
    }
    
}

void loop() {
}
'''

earlephilhower, 2019-08-21:
Closing this for now.  OTA reported working fine in the latest release, and the only outstanding bit is the http.GET method which is still borked (but with a PR in-progress).  We'll track that problem in the consolidated issue referenced above.

closed 2019-08-21 by earlephilhower