esp8266/Arduino #5823: configTime / sntp_set_timezone supports only full hour timezones (+/- 30min) timezones not supported

jyberg, 2019-02-26:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|Esp8266|NodeMcu]
- Core Version:			[2.20500.0]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Nodemcu|]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

configTime function (Time.c) supports timezone definition is given in seconds. But  sntp_set_timezone(sint8 timezone) in sntp-lwip2.c requires time zone in hours. This is fault because there exsist other timezones then full hours also.

Please change sntp_set_timezone function to accept second based timezone, and validate zone using second values if(timezone >= -11*3600 || timezone <= 13*3600)

Dayligt savint time setup sntp_set_daylight(...) seems to use second values correctrly

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

// Setup Wifi connection

// Set TimeZone 5h 30min
int timeZoneH 5;
int timeZoneM 30;
	configTime((timeZoneH*60+timeZoneM)*60, 0, "pool.ntp.org", "time.nist.gov");
	time(nullptr);
}

void loop() {
		time_t now = time(nullptr);
		tm* t = localtime(&now);
// returned time is not in correct timezone 5h30min...

}

oddstr13, 2019-02-27:
[List of UTC time offsets](https://en.wikipedia.org/wiki/List_of_UTC_time_offsets)

I would like to point out that a few are :45 off the full hour, and that historically there has been instances of higher precision than 15min, for example [+01:24](https://en.wikipedia.org/wiki/UTC%2B01:24), [+04:51](https://en.wikipedia.org/wiki/UTC%2B04:51) and [-00:25:21 (second precision no less)](https://en.wikipedia.org/wiki/UTC%E2%88%9200:25:21).
None of the odd non-15min increments are in use today as far as I can tell, but it would probably be best  with at least minute precision for time zones.

jyberg, 2019-02-27:
> [List of UTC time offsets](https://en.wikipedia.org/wiki/List_of_UTC_time_offsets)
>
> I would like to point out that a few are :45 off the full hour, and that historically there has been instances of [...]

yes that why I suggested to use seconds value like in configTime interface function uses and not convert to hours/minutes

devyte, 2019-02-28:
Please test #5828 , it adds sntp_set_timezone_in_seconds(). You may have to 
`extern sntp_set_timezone_in_seconds(sint32 timezone);`
at the top of your sketch, because the function signature isn't in a .h yet. Also, configTime() isn't updated either.

PR #5828 (2019-03-01): Implement sntp_get|set_timezone_in_seconds()
Fixes #5823
closed 2019-03-01 by devyte