esp8266/Arduino #5830: [2.6.0-dev] Fatal error / crash in SPIFFS

cziter15, 2019-02-28:
- Hardware:					ESP-12
- Core Version:			2.6.0-dev ead83c16a60793d914a3370d46dd8e22d98e107a
- Development Env:		Arduino IDE/Visual Micro
- Operating System:		Windows

### Settings in IDE

- Module:  			ESPIno 12E
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		doesn't matter
- Upload Speed:   115200

### Problem Description

Lastest git crashes at SPIFFS open/exist. There was no problem in 2.5.0 release.

### [MCVE]

```cpp

#include <Arduino.h>

void setup() {
        if (SPIFFS.begin())
        { 
             if (File f = SPIFFS.open("/settings.dat")
             {
                 f.close();
             }
        }
}

void loop() {

}

```

### Debug Messages

0x4020b148: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at esp8266\cores\esp8266\spiffs_hal.cpp line 45
0x4020b204: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at esp8266\2.5.0\cores\esp8266\spiffs_hal.cpp line 102
0x4020b340: spiffs_hal_erase(unsigned int, unsigned int) at esp8266\cores\esp8266\spiffs_hal.cpp line 168
0x4020ac60: SPIFFSImpl::open(char const*, fs::OpenMode, fs::AccessMode) at

earlephilhower, 2019-02-28:
The MCVE given doesn't compile.

````
/home/earle/Arduino/sketch_feb28b/sketch_feb28b.ino: In function 'void setup()':
sketch_feb28b:6:45: error: no matching function for call to 'fs::FS::open(const char [14])'
     if (File f = SPIFFS.open("/settings.dat")) {
                                             ^
/home/earle/Arduino/sketch_feb28b/sketch_feb28b.ino:6:45: note: candidates are:
In file included from /home/earle/Arduino/sketch_feb28b/sketch_feb28b.ino:2:0:
/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/FS.h:166:10: note: fs::File fs::FS::open(const char*, const char*)
     File open(const char* path, const char* mode);
          ^
/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/FS.h:166:10: note:   candidate expects 2 arguments, 1 provided
/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/FS.h:167:10: note: fs::File fs::FS::open(const String&, const char*)
     File open(const String& path, const char* mode);
          ^
/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/FS.h:167:10: note:   candidate expects 2 arguments, 1 provided
exit status 1
no matching function for call to 'fs::FS::open(const char [14])'
````

cziter15, 2019-02-28:
I'm sorry, I didn't verify this properly.
Looks like SPIFFS.open stopped accepting constant char arrays with PROGMEM specifier.
Removing PROGMEM fixes issue.

earlephilhower, 2019-03-01:
Thanks for the debug.  I don't think SPIFFS filenames ever officially supported being in PROGMEM (the pointer is passed directly into the SPIFFS library which knows *nothing* about `pgm_read_byte` and assorted macros.

Closing.

closed 2019-03-01 by earlephilhower