esp8266/Arduino #5832: 2.6.0 is significantly faster than 2.5.0

scottchiefbaker, 2019-02-28:
I don't have an issue, I just wanted to say Good Job (TM) to the team working on this firmware!

I'm using this [SHA256](https://gist.github.com/scottchiefbaker/641f69eb64bbda2809c03300c58c3266) script to benchmark my ESP8226. Comparing the results between 2.5.0 and 2.6.0-dev I'm seeing significant speed improvements:

 x | 2.50 | 2.6.0-dev | Percent Improvement
------|-----|-----|--
CPU @ 80Mh | ~9520 iterations | ~11010 iterations | 15.6%
CPU @ 160Mh | ~17270 iterations | ~22110 iterations | 28.0%

The **exact** same code, running on the **exact** same hardware is showing a 15% improvement. When you bump the CPU up to 160Mhz the improvements are even more impressive! Excellent job.

Since this really isn't an issue I'll leave it open for a couple of days so people can see it and then I'll close it.

cziter15, 2019-02-28:
Yes, that is caused by ESP SDK revert.

TD-er, 2019-02-28:
Indeed, the default is now SDK2.2.1 again.
Core 2.5.0 is using SDK 3.0.0-dev

See also https://github.com/esp8266/Arduino/issues/5513#issuecomment-465085062

devyte, 2019-03-01:
Having our effort appreciated is rare and always encouraging, but previous comments are correct: it's the SDK that was reverted, and not really our efforts. Please see pinned issue #5784.
Closing.

closed 2019-03-01 by devyte