esp8266/Arduino #5833: Question: Using BearSSL for STARTTLS in SMTP

michaelradhuber, 2019-03-01:
This is a question directed primarily to @earlephilhower  - would appreciate a short response:

I have been trying to tweak the ESP8266 BearSSL library to get to work STARTTLS for sending emails via SMTP / STARTTLS servers. Basically I connect directly to the mailserver with the normal wificlient. After the STARTTLS command was accepted with 220 I directly call the (private) function from the WiFiClientSecure class _connectSSL (taking advantage of the already existing TCP/IP sock:

`clientsecure._connectSSL(this->smtp_server);`

In the WiFiClientSecureBearSSL.cpp I get as far as this func:

```
_bool WiFiClientSecure::_wait_for_handshake() {
  _handshake_done = false;
  Serial.println("Entering handshake loop -wait-fot-handshake func");
  while (!_handshake_done && _clientConnected()) {
    int ret = _run_until(BR_SSL_SENDAPP);
    Serial.println(" ** ");
    if (ret < 0) {
      Serial.println("_wait_for_handshake: failed\n");
      break;
    }
    if (br_ssl_engine_current_state(_eng) & BR_SSL_SENDAPP) {
      _handshake_done = true;
    }
    optimistic_yield(1000);
  }
  return _handshake_done;
}
```

...where I get stuck in the while loop. I admit not having yet digged deep enough into the BearSSL library to get the meaning of the BR_SSL_SENDAPP,

my mail server log shows: 

```
SSL_accept error from ...: Connection timed out
lost connection after STARTTLS from ...
```

This is the (messy) code I have been messing around with to call BearSSL:

```
#include <EEPROM.h>
#include <TimeLib.h>
#include <WiFiUdp.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <EMailSender.h>
#include "Arduino.h"

#define DEBUG_SSL
#define DEBUGV
#define HEAP_DEBUG
#define DEBUG_ESP_SSL //define DEBUG

static const char ntpServerName[] = "at.pool.ntp.org";
const int timeZone = 1;
WiFiUDP Udp;
unsigned int localPort = 8888; // local port to listen for UDP packets

time_t getNtpTime();
void sendNTPpacket(IPAddress &address);

/*-------- NTP code ----------*/

const int NTP_PACKET_SIZE = 48; // NTP time is in the first 48 bytes of message
byte packetBuffer[NTP_PACKET_SIZE]; //buffer to hold incoming & outgoing packets

time_t getNtpTime()
{
  IPAddress ntpServerIP; // NTP server's ip address

[... 222 lines cut ...]

And here is the specific function from the EMailSender library where I fail:

```
 DEBUG_PRINTLN(F("STARTTLS:"));
  client.println(F("STARTTLS"));

  response = awaitSMTPResponse(client, "220", "Ready to start TLS");
  Serial.println("Response set");
  if (!response.status) return response;
  Serial.println("Starting TLS handshake now");
  bool sslstatus = clientsecure._connectSSL(this->smtp_server);
  while (!sslstatus) {
    Serial.print(" . ");
    delay(500);
  }
  if (sslstatus) DEBUG_PRINTLN(F("SSL/TLS connection established successfully!"));
  if (!sslstatus) {
    DEBUG_PRINTLN(F("SSL/TLS connection ERROR!"));
    DEBUG_PRINT(F("Get last SSL error: "));
    DEBUG_PRINTLN(clientsecure.getLastSSLError());
  }

Thank you in advance!
```

devyte, 2019-03-01:
As stated, not an issue, but a question. Closing.

closed 2019-03-01 by devyte

earlephilhower, 2019-03-01:
@michaelradhuber, @devyte's correct but this is an interesting use case.

Your best bet for debugging this is to check out https://bearssl.org and build an app that uses plain TCP connections on your machine implementing the handoff.  You can add printfs() throughout the ssl code to figure out where things are going pear shaped.  That SENDAPP bit is the flag that BearSSL is ready for unecrypted application data (i.e. it's done handshake).  It's documented at the main site pretty well, but it does take a bit to get the underlying design as it's got Pseudo-Forth and state machines...

I'm not familiar with the TLS extensions to SMTP, but there may be a different handshake sequence vs. a TLS handshake starting at the first byte transmitted.

Also there could be bad stuff going on since having two separate WiFiClient and WiFiClientSecure with the same TCP connection (which is what you end up if you try and sneak the actual LWIP socket from an existing, working WFC into a brand new WFCS).

michaelradhuber, 2019-03-02:
@earlephilhower thanks for this response. As to your last point, my bet is that this is what's actually causing me most troubles. I'm still working on it, but calling a TCP connection from another context (out of this class) most surely won't work. So I'm trying to expand the WiFiClientSecureBearSSL by hereditation, to split up the connect() function into the normal TCP sock connection (WiFiCLient), and subsequently a second function for running _connectSSL().
Then I guess I will also need to rewrite the buffer read functions for reading normal text without encryption.
I don't want to dig too deep into the BearSSL class for this, also my knowledge on Forth is actually zero (might want to change this, but not right now).
According to the RFC for STARTTLS [https://www.ietf.org/rfc/rfc3207.txt](url), the TLS protocoll is actually the same. So the handshake sequence for TLS [https://tools.ietf.org/html/rfc5246](url) SHOULD actually be identical (if not, I will drop this idea, but for the moment I have to work on this assumption). Therefore I would like to end up with my own class for STARTTLS, based on the underlying Arduino and BearSSL implementations, without changing any of that.
If you ahve any more hints, let me know. For the while being, thanks for the input :)
Mike