esp8266/Arduino #5834: upload_port setting in platformio.ini seems broken in 2.0.x for serial ports

dalbert2, 2019-03-01:
Sorry if I'm missing something, but the new 2.0.x releases appear to break the upload_port setting in platformio.ini: it looks like it now interprets the upload_port as an IP address instead of a COM port (at least on windows platforms), so for example:
```
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
upload_port=COM5
```
Now results in:
```
Uploading .pioenvs\esp12e\firmware.bin
12:08:00 [DEBUG]: Options: {'timeout': 10, 'esp_ip': 'COM5', 'host_port': 46511, 'image': '.pioenvs\\esp12e\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
12:08:00 [INFO]: Starting on 0.0.0.0:46511
12:08:00 [INFO]: Upload size: 262160
Sending invitation to COM5 ..........
12:09:41 [ERROR]: No response from the ESP
```

Any ideas?  I didn't see anything helpful in the release notes.  How do I tell it to use a specific serial port for upload?  (Note: if no upload port is specified, it auto-detects the first serial port, but that's not the one I need to use).

Also suggestion: when creating a new Arduino ESP8266 project, it would be very helpful for new users to include  a comment in the auto-created platformio.ini referencing this page: 
https://docs.platformio.org/en/latest/platforms/espressif8266.html#configuration

earlephilhower, 2019-03-01:
@dalbert2, you need to open this in the Platform.IO tracker.  This is the Arduino core, and PIO does not use our build or upload methods.

closed 2019-03-01 by earlephilhower

dalbert2, 2019-03-01:
Will do, sorry for posting in the wrong spot and thanks for the pointer!