esp8266/Arduino #5836: name of wifi libraries

aster94, 2019-03-02:
as you know better then me most of the esp8266 wifi library is compatible with the [official arduino wifi class](https://www.arduino.cc/en/Reference/WiFi). The ESP32 core uses also the same name for the libraries (i.e: [WiFi.h](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi))

Considering that at this point renaming the `ESP8266WiFi.h` to `WiFi.h` for 100% compatibility woudl break almost 100% of the sketches would it be possible to make a `WiFi.h` which points to `ESP8266WiFi.h`?

devyte, 2019-03-02:
There are two possible approaches:
1. Make the current WiFi fully compatible.
This requires investigating the differences, figuring out what is missing, and figuring out what is not compatible. If no conflicts are found, overload methods would need to be implemented for whatever is missing or different, so that the currnt ESP8266WiFi becomes a superset of the Arduno official WiFi. Finally, a new WiFI.h would need to be implemented that typedefs ESP8266WiFI to WiFi.
2. Encapsulation
This would implement a new class WiFi that contains an ESP8266WiFi as a member, and implements forwarder methods only for the official Arduino WiFi api. This assumes that there are no behavioral conflicts.

In either case, effort is required that won't be provided by the current repo maintainers, because there are much higher priority issues to address, and a lot of them at that.

If you wish to put forth the effort for this yourself, please make a proposal in a PR, and we can discuss further there. In that case, please reference this issue.

Closing as won't fix for now.

closed 2019-03-02 by devyte

aster94, 2019-03-02:
I understand, maybe when one day a big rework of the code will be needed the renaming could be taken in consideration. It is nice to have libraries with the same purpose/API also with the same name along different architectures