esp8266/Arduino #5838: delay(1) in library sometimes causes exception(9)

mamama1, 2019-03-03:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[NodeMCU 4M]
- Core Version:			[2.5, 2.4]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[nodemcuv2]
- Flash Mode:			[noidea]
- Flash Size: 		[4MB]
- lwip Variant:   [default?]
- Reset Method:		[nodemcu i guess]
- Flash Frequency:	[40Mhz i guess]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [512000 (CP2102) (serial upload only)

### Problem Description

I'm using this library for EEPROM access: https://github.com/JChristensen/extEEPROM
and it seems this thing is kind of incompatible with the ESP8266. See here, I have already figured some things out: https://github.com/JChristensen/extEEPROM/issues/19

From then on, I only sometimes got crashes but they were still there. recently they started to annoy me so I tracked them down to delay(1) which is called in my customized version of extEEPROM.cpp:

```cpp
        //wait up to 50ms for the write to complete
        for (uint8_t i=50; i; --i) {
            ESP.wdtFeed();
            //delay(1);                     //no point in waiting too fast
            communication->beginTransmission(ctrlByte);
            if (_nAddrBytes == 2) communication->write((byte)0);        //high addr byte
            communication->write((byte)0);                              //low addr byte
            txStatus = communication->endTransmission();
            if (txStatus == 0) break;
        }
```
(original file, see here: https://github.com/JChristensen/extEEPROM/blob/master/extEEPROM.cpp starting at line 134)

When I removed delay(1), the crashes disappeared.
This is, how I was writing data to the EEPROM:

```cpp
        typedef struct __attribute((__packed__)) NodePairing
            {
                uint8_t nodeID;
                bool active;
                uint16_t type;
                uint8_t serialNumber[NODE_SERIAL_LEN];
                uint8_t name[20];
            } NodePairing_t;

            NodePairing_t nodesPairing[NODE_COUNT_MAX]; // (about 8k)

extEEPROM.write(EEPROM_NODE_PAIRINGDATA_OFFSET, reinterpret_cast<byte*>(&nodesPairing), (size_t)sizeof(nodesPairing));
```

this is the stack trace i was receiving:

> Exception (9):
[... 489 lines cut ...]
> 0x40229206: ip4_output_if_src at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 833
> 0x40229d4b: ip_chksum_pseudo at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c line 542
> 0x402251b8: udp_sendto_if_src at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 1049
> 0x40101300: malloc at C:\users\user\appdata\roaming\spb_data\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.c line 1673
> 0x4022520f: udp_sendto_if at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 1049
> 0x40105a5a: wdt_feed at ?? line ?
> 0x402252a8: udp_sendto at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 1049
> 0x40105b16: ets_timer_setfn at ?? line ?
> 0x40105ce8: ets_timer_arm_new at ?? line ?
> 0x402157e4: esp_yield at C:\users\user\appdata\roaming\spb_data\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 91
> 0x40201d4f: delay at C:\users\user\appdata\roaming\spb_data\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.c line 51
> 0x40213609: aaaNTPClient::forceUpdate() at c:\PlatformIO_Projects\aaaHC5_Gateway/lib\aaaNTPClient/aaaNTPClient.cpp line 50
> 0x402136a7: aaaNTPClient::update() at c:\PlatformIO_Projects\aaaHC5_Gateway/lib\aaaNTPClient/aaaNTPClient.cpp line 50
> 0x40209b62: loop at c:\PlatformIO_Projects\aaaHC5_Gateway/src/main.cpp line 567
> 0x4020b070: onWsEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at c:\PlatformIO_Projects\aaaHC5_Gateway/src/main.cpp line 635
> 0x40217b94: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\user\appdata\roaming\spb_data\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1934
> 0x40217b74: std::_Function_handler ::_M_invoke(std::_Any_data const&, AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at c:\users\user\appdata\roaming\spb_data\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
> 0x40215805: esp_schedule at C:\users\user\appdata\roaming\spb_data\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 95
> 0x40215890: loop_wrapper at C:\users\user\appdata\roaming\spb_data\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 125
> 

is this somehow related to https://github.com/esp8266/Arduino/issues/5722 or to "trying to use yield/delay, or libs that use yield/delay, from inside async callbacks"?
This is not an async callback...

Thanks!

devyte, 2019-03-03:
Are you doing eeprom ops from an async server callback?

mamama1, 2019-03-03:
lol well, yeah. never put it that way but as you're pointing it out now, and as I have just closed all VS Code windows and released the pressure from my brain, it is completely obvious.

may I slap myself now? :-D

won't this EVER work reliably or will this only be a problem if it takes too long?

is an async callback always an interrupt where I only should set flags?

devyte, 2019-03-04:
Closing due to user error.
Ref [this](https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#other-causes-for-crashes) .

Async callbacks exexute in the SYS context, and there are restrictions that apply. In general, blocking ops are a big NO, unless you know exactly what you're doing.
I suggest looking at scheduling your function, e.g. with Ticker. The scheduled api calls your function in the CONT context, as though it were part of the loop. In that case, you can yield/delay/etc.

closed 2019-03-04 by devyte

mamama1, 2019-03-04:
i have now made a function processCallbacks() which is called in the loop and which looks in a vector of a "callback" called struct which contains stuff to do.
the vector is populated from the async callback. so far, no crashes. stupid me.

devyte, 2019-03-04:
You're reinventing the wheel, that's more or less what the scheduled functions do: when you schedule a function, it gets added to a container, then all functions in the container get called together with the next loop. Look at the scheduled functions api, I think that using it will simplify your code.