esp8266/Arduino #5839: WiFi.config(fixedIP, fixedGateway, fixedNetmask) disconnects ESP on 2.5.0 (regression from 2.4.2)

newHeiko, 2019-03-03:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it. (no stack dump)
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Slackware Linux]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[unknown]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

When changing the IP address configuration with WiFi.config after connecting, 2.5.0 just releases the DHCP lease and returns the IP address but does not enable the fixed IP address. This works on 2.4.2.

After a disconnect->reconnect cycle, 2.5.0 activates the fixed IP address. (Side note: This seems shaky on 2.4.2 - does not work every time)

I checked the github web interface but could not make out any relevant changes to the ESP8266WiFi tree since 2.5.0, so I hope it's OK that I did not check latest-git.

If you need any more info, just go ahead and ask :)

Heiko

May be related to #5757 ?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  delay(100);

  IPAddress fixedIP(192, 168, 205, 225);
  IPAddress fixedGateway(192, 168, 205, 250);
  IPAddress fixedNetmask(255, 255, 255, 0);

  WiFi.config(0u, 0u, 0u);
  WiFi.begin("ssid", "psk");

  delay(10000);

  Serial.println("Switching to fixed IP - this is where 2.4.2 starts to respond to pings to .225, 2.5.0 does not");
  WiFi.config(fixedIP, fixedGateway, fixedNetmask);

  delay(5000);

  Serial.println("Reconnecting");
  WiFi.disconnect();
  delay(1000);
  WiFi.begin("ssid", "psk");
  Serial.println("Now 2.5.0 responds to pings to .225");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
### Debug Messages on 2.5.0
```
0x2d
csum 0x2d
v951aeffa
~ld
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with jmri-american, channel 1
dhcp client start...
ip:192.168.205.228,mask:255.255.255.0,gw:192.168.205.250
Switching to fixed IP - this is where 2.4.2 starts to respond to pings to .225, 2.5.0 does not
pm open,type:2 0
Reconnecting
state: 5 -> 0 (0)
rm 0
pm close 7
Now 2.5.0 responds to pings to .225
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with jmri-american, channel 1
ip:192.168.205.225,mask:255.255.255.0,gw:192.168.205.250
ip:192.168.205.225,mask:255.255.255.0,gw:192.168.205.250
pm open,type:2 0
```
### Debug Messages on 2.4.2
```
0x2d
csum 0x2d
vbb28d4a3
~ld
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with jmri-american, channel 1
dhcp client start...
ip:192.168.205.228,mask:255.255.255.0,gw:192.168.205.250
Switching to fixed IP - this is where 2.4.2 starts to respond to pings to .225, 2.5.0 does not
ip:192.168.205.225,mask:255.255.255.0,gw:192.168.205.250
pm open,type:2 0
Reconnecting
state: 5 -> 0 (0)
rm 0
pm close 7
Now 2.5.0 responds to pings to .225
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with jmri-american, channel 1
ip:192.168.205.225,mask:255.255.255.0,gw:192.168.205.250
ip:192.168.205.225,mask:255.255.255.0,gw:192.168.205.250
pm open,type:2 0
```

kapyaar, 2019-03-04:
Hmm. I think I can second this finding. 

Platform
Hardware:	[ESP-12]
Core Version:	[2.5.0]
Development Env:	[Arduino IDE]
Operating System:	[Windows]

Settings in IDE
Module: [Nodemcu]
Flash Mode:	[DOUT]
Flash Size: [4MB]
lwip Variant: [v2 Lower Memory]
Reset Method:	[nodemcu]
Flash Frequency:	[40Mhz]
CPU Frequency:	[80Mhz]
Upload Using:	[SERIAL]
Upload Speed: [921600] (serial upload only)

I simply used BasicHTTPClient example, with and without 

```
  IPAddress fixedIP(192, 168, 0, 219);
  IPAddress fixedGateway(192, 168, 0,1);
  IPAddress fixedNetmask(255, 255, 255, 0);
  WiFi.config(fixedIP, fixedGateway, fixedNetmask);
```
 
after 

```
WiFi.mode(WIFI_STA);
WiFiMulti.addAP("SSID", "PASSWORD");
```

With Static IP Settings, I get 

```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
```

With DHCP ( ie , Static IP settings commented out) I get 

```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
<HTML>
<HEAD>
  <!-- Created with AOLpress/2.0 -->
  <TITLE>Connection Header</TITLE>
</HEAD>
<BODY>
```
......... and the rest of the html

d-a-v, 2019-03-04:
`WiFi.config()` is supposed to be called before `WiFi.begin()` right ?
Doing so, it works (then `WiFi.config(0u,0u,0u)+WiFi.begin()` restores use of dhcp if needed).

However:
- I can reproduce the issue with the above MCVE and master
- I have not tested with 2.4.2, but with lwip-v1.4 instead of lwIP-v2, it works with master (no issue)
- [documentation](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#config) places `.config()` *after* `.begin()` which is counter-productive (dhcp client is started for nothing)

So we must fix documentation, and maybe fix this should-not-be-documented regression.

newHeiko, 2019-03-04:
Thanks for confirming and testing with lwip-v1.4. I can confirm it works with lwip-v1.4 on 2.5.0, so I have a way forward, which is great!

Simply adding WiFi.begin(); after WiFi.config(fixedIP, fixedGateway, fixedNetmask); does not work, I need to do a WiFi.disconnect() in between. But IMHO, there is a use case for on-the-fly IP reconfiguration without reconnecting to the WiFi AP.

Like on my system: A couple of ESP-clients run with DHCP addresses, but when I want to change the configuration of one of them, I go there, push a button, let the ESP replace the last octet of the IP address with a fixed one so I can connect to a webserver running on the ESP. It is a pain to have the device reconnect to the WiFi every time.

Anyway, if that's what I have to do, that's what I have to do ;)

Heiko

kapyaar, 2019-03-05:
Yes, confirming 1.4 working. As to the order of wifi.config and wifi.begin, I have been using config after begin for a long time :) I suppose that though it makes sense that the correct order is config before begin, it still works?

dmdt, 2019-03-09:
I also have problem with WiFi.config(). But it appears time to time. Some time all working great. I implement new function and softAP doesn't start with WiFi.config(). Comment this line - all works great. Wrote one more function's (it doesn't work with wifi) and softAP again doesn't start. When I uncomment - all ok. And so on.

kapyaar, 2019-04-03:
@d-a-v Just a follow up comment. Going back to v1.4 resolves the static IP issue. But the v1.4 has the issue of "LmacRxBlk:1" being thrown in the terminal. I see that you added a 'bug' label added to this issue. I hope the static IP issue on v2 will be resolved soon.

d-a-v, 2019-04-30:
> "LmacRxBlk:1"

Tell me about it. It was the primary reason of lwip2 existence.

> I hope the static IP issue on v2 will be resolved soon.

Nothing prevents to move `WiFi.config()` before `WiFi.begin()`.
However, fix is coming with #6026.

PR #6026 (2019-05-01): fix switching to static address with lwip2
For some reason, ip address is not propagated in a visible way for lwip2
when switching to static address (wifi.config()) *after* wifi.begin().

This patch calls lwip-v1.4's netif_set_addr() with the new ip address to set
all things up, just like it is done and right when wifi.begin() is called
after wifi.config().

Also tested when IPv6 is enabled.

fixes #5839
fixes #6024
commit 17500226 (2019-05-01): fix switching to static address with lwip2 (#6026)

fix switching to static address with lwip2

For some reason, ip address is not propagated in a visible way for lwip2
when switching to static address (wifi.config()) *after* wifi.begin().

This patch calls lwip-v1.4's netif_set_addr() with the new ip address to set
all things up, just like it is done and right when wifi.begin() is called
after wifi.config().

Also tested when IPv6 is enabled.

fix documentation: It is more natural to set an IP address before starting WiFi
(.. and not after dhcp has started)
fixes #5839
fixes #6024
closed 2019-05-01 by d-a-v