esp8266/Arduino #5842: STR(ARDUINO_ESP8266_GIT_DESC) results in "unspecified"

luffykesh, 2019-03-05:
I am using the git version.
SDK: 2.2.1

- When a sketch is compiled for the first time
`STR(ARDUINO_ESP8266_GIT_DESC)` renders to "unspecified"

- Then if the sketch is recompiled again
`STR(ARDUINO_ESP8266_GIT_DESC)` results in a appropriate string "2.5.0-42-g82da5c76"

AyushMarsian, 2019-03-07:
I am using windows machine with windows8.1 OS and git version.
SDK: 2.2.1

I am also facing the same issue.

earlephilhower, 2019-11-18:
@d-a-v, didn't you push a fix for this a while back?

d-a-v, 2019-11-18:
@luffykesh @AyushMarsian Is this still true with 2.6.1 or current git ?

luffykesh, 2019-11-19:
Will try it out and report today.

earlephilhower, 2020-01-01:
No update in a couple months.  I'm pretty sure @d-a-v fixed this a while back.  If it's still a problem, please open a new issue and we'll look at it.

closed 2020-01-01 by earlephilhower