esp8266/Arduino #5848: WiFi.begin() dont connect in 2.4.2 and lower

MrTilllus, 2019-03-07:
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP8285 device
- Core Version:			2.4.2, 2.4.0, 2.5.0
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu 0.9 ESP-12
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory|Higher Bandwidth
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

If I want to connect on core 2.5 its no problem, but I want to use a Lib that has some issues with the 2.5 so i tried 2.4.2 and 2.4 and there I cant connect to my Wifi, there are no Errors or Warnings but when i upload my sketch ist just printing points. Here is my Code:

#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.println();

  WiFi.disconnect(true);
  delay(20);
  WiFi.mode(WIFI_STA);
  delay(20);
  
  WiFi.begin("SSID", "PW");    

  Serial.println( WiFi.macAddress() );
  
  Serial.print("Connecting");

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println();
  Serial.print("Connected, IP address: ");
  Serial.println( WiFi.macAddress() );
  Serial.println( WiFi.localIP() );
  Serial.println( ESP.getChipId() );
  
}

void loop() {
  
}

d-a-v, 2019-03-07:
Can you try with "erase flash: all flash contents" tool menu option once when you update your core ?

MrTilllus, 2019-03-07:
i tried befor and it changed nothing

d-a-v, 2019-03-07:
If it works with 2.5.0, then it should also work with 2.4.2.
Try to uninstall and remove everything (including the `arduino15/` direcory) and try again with 2.4.2.
https://forum.arduino.cc/index.php?topic=362507.0

MrTilllus, 2019-03-10:
it changed nothing

earlephilhower, 2019-07-25:
@d-a-v, any objections to closing this?  We're not following a branched development model so there will not be any changes to the 2.4.2 release.

earlephilhower, 2020-02-24:
Closing as this is not a problem in 2.5.0 and above.

closed 2020-02-24 by earlephilhower