esp8266/Arduino #5852: PlatformIO get error at ArduinoOTA

Findus76, 2019-03-08:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[see below]
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [NA] (serial upload only)

### Problem Description

Detailed problem description goes here.

Hello
My compilation environment has stopped working for no reason.
With the same project set up on another computer, the compilation works.
Earlier projects can also no longer be comiled.

**Error code from compilation**
`Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\ESP8266WiFiSTA.cpp.o

Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\WiFiServer.cpp.o
Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
Compiling .pioenvs\esp12e\lib6b8\ESP8266WiFi\WiFiUdp.cpp.o
Compiling .pioenvs\esp12e\lib19b\ArduinoOTA_ID6178\InternalStorage.cpp.o
Compiling .pioenvs\esp12e\lib19b\ArduinoOTA_ID6178\InternalStorageAVR.cpp.o
Compiling .pioenvs\esp12e\lib19b\ArduinoOTA_ID6178\OTAStorage.cpp.o
.piolibdeps\ArduinoOTA_ID6178\src\InternalStorage.cpp:43:21: warning: 'long_call' attribute directive ignored [-Wattributes]
void waitForReady() {
^
.piolibdeps\ArduinoOTA_ID6178\src\InternalStorage.cpp:52:63: warning: 'long_call' attribute directive ignored [-Wattributes]
static void eraseFlash(int address, int length, int pageSize)
^
.piolibdeps\ArduinoOTA_ID6178\src\InternalStorage.cpp:82:76: warning: 'long_call' attribute directive ignored [-Wattributes]
static void copyFlashAndReset(int dest, int src, int length, int pageSize)
^
.piolibdeps\ArduinoOTA_ID6178\src\InternalStorage.cpp: In function 'void copyFlashAndReset(int, int, int, int)':
.piolibdeps\ArduinoOTA_ID6178\src\InternalStorage.cpp:95:22: error: 'NVIC_SystemReset' was not declared in this scope
NVIC_SystemReset();
^
Compiling .pioenvs\esp12e\lib19b\ArduinoOTA_ID6178\WiFiOTA.cpp.o
*** [.pioenvs\esp12e\lib19b\ArduinoOTA_ID6178\InternalStorage.cpp.o] Error 1
.piolibdeps\ArduinoOTA_ID6178\src\OTAStorage.cpp: In constructor 'OTAStorage::OTAStorage()':
.piolibdeps\ArduinoOTA_ID6178\src\OTAStorage.cpp:32:1: error: expected identifier before '{' token
{
^
[... 253 lines cut ...]
serialiosoftwareserial
Reveal
TFT_Touch_Shield_V2
 0.0.0
uncategorized
Reveal
Ticker by Ivan Grokhtokov <ivan@esp8266.com>
 1.0
Allows to call functions with a given interval.
timing
Reveal
Wire by Arduino
 1.0
Allows the communication between devices or sensors connected via Two Wire Interface Bus. For esp8266 boards.
signalinputoutput
Reveal
ESP8266 by Simon Peter,Markus Sattler,Ivan Grokhotkov
 1.0
ESP8266 sketches examples
other
Reveal
Web·Open Source·Get Started·Docs·News·Community·Contact Us`

Brg
Findus

jandrassy, 2019-03-08:
you have the non esp ArduinoOTA library. the ArduinoOTA library in esp8266 core package should have precedence

Findus76, 2019-03-08:
Many thanks!
My weekend is saved!

Brg
Findus

closed 2019-03-08 by Findus76