esp8266/Arduino #5855: OTA crash with UDP server and CORE debug

luffykesh, 2019-03-09:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
-  [   ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			git | last commit I have  `2528013c5dc1ce7703bf211b41a3aefcd6afe0e5`
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:	ck
- Flash Frequency:	40Mhz
- CPU Frequency:   80Mhz
- Upload Using:       OTA
- Debug Port:          Serial
- Debug Level:        Core
- Espressif FW:         2.2.1

### Problem Description

Detailed problem description goes here.

OTA updates crash with `Fatal exception 3(LoadStoreErrorCause)` when there is a UDP server running **AND** CORE debug is enabled.
If CORE debug is disabled, it works just fine.

Haven't been able to decode the stack, because with the the git version the ESPExceptionDecoder is not able to locate the file `xtensa-lx106-elf-addr2line`:`ERROR: xtensa-lx106-elf-addr2line not found!`

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
#include <WiFiUDP.h>

unsigned int localPort = 8888;

#ifndef STASSID
#define STASSID "ssid"
#define STAPSK  "password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;
WiFiUDP Udp;
char packetBuffer[UDP_TX_PACKET_MAX_SIZE];

void setup(void) {
  Serial.begin(9600);
  Serial.println();
  Serial.println("Booting Sketch...");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }
  Serial.print("IP: ");
  Serial.println(WiFi.localIP());
  httpUpdater.setup(&httpServer);
  httpServer.begin();
  Udp.begin(localPort);

}
void loop(void) {
  httpServer.handleClient();
    // if there's data available, read a packet
  int packetSize = Udp.parsePacket();
  if (packetSize) {
    Serial.print("Received packet of size ");
    Serial.println(packetSize);
    Serial.print("From ");
    IPAddress remote = Udp.remoteIP();
    for (int i = 0; i < 4; i++) {
      Serial.print(remote[i], DEC);
      if (i < 3) {
        Serial.print(".");
      }
    }
    Serial.print(", port ");
    Serial.println(Udp.remotePort());

    // read the packet into packetBufffer
    Udp.read(packetBuffer, UDP_TX_PACKET_MAX_SIZE);
    Serial.println("Contents:");
    Serial.println(packetBuffer);
  }
	delay(10);
}
```
### Debug Messages
Attaching the stack trace.
```
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4024496b, depc=0x00000000

Exception (3):
epc1=0x4000228b epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024496b depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb00 end: 3fffffc0 offset: 01a0
3ffffca0:  4000050c 40104284 3ffec4d0 4020a438  
3ffffcb0:  0000002c 00000030 3fff0930 3fff0930  
3ffffcc0:  3fff1eea 00000000 0000002c 4020a626  
3ffffcd0:  0000001c 40207738 3fff0930 3fff0930  
3ffffce0:  00000000 3ffffd50 3ffffd50 3ffe879a  
3ffffcf0:  40002514 3fffdd3c 3fff0750 3fff2cc4  
3ffffd00:  00000001 00000020 00000008 00000000  
3ffffd10:  00000000 00000000 00000000 4024496b  
3ffffd20:  3fff1f4c 00000000 0000002c 3ffffde0  
3ffffd30:  3ffffd50 3ffffd50 00000004 40202c58  
3ffffd40:  3fff0608 00000015 3ffffd80 00000000  
3ffffd50:  3ffe8798 4024496b 3fff0608 3fff0608  
3ffffd60:  00ff0000 40404040 3fff1944 3ffffde0  
3ffffd70:  3fff2cc4 3fff0620 3fff1944 40206097  
3ffffd80:  00000000 00000000 ff000000 402076c0  
3ffffd90:  3fff0694 3ffffde0 3ffffdc0 3ffffde0  
3ffffda0:  3fff0694 00000001 3fff16b4 40204466  
3ffffdb0:  3fff0694 40202cb8 3fff16b4 402044e1  
3ffffdc0:  00000000 00000000 ffff0650 402079e4  
3ffffdd0:  3fff0694 3fff16b4 3fff0650 402051af  
3ffffde0:  6470752f 00657461 00707954 00000000  
3ffffdf0:  00000000 ffff21e0 3fff2324 001d001f  
3ffffe00:  ff00000d 3fff1f4c 0015001f ff006e69  
3ffffe10:  00000000 00000000 00ffffff 3fff22a4  
3ffffe20:  0006003f ffff21e0 3fff1edc 0023005f  
3ffffe30:  ffffffff 3fff21e0 3fff0650 40204ab5  
3ffffe40:  3fff0678 3ffffeb0 00000000 40204c54  
3ffffe50:  3fff21c8 0004b94e 402044b4 40202cb8  
3ffffe60:  3ffe8ae3 3ffffe98 3fff0650 0004b94e  
3ffffe70:  3ffe8ae3 3ffffe98 3fff0650 40205b2d  
3ffffe80:  00000000 00000000 00736e49 00000000  
3ffffe90:  00000000 ff000001 3fff1dec 0001005f  
3ffffea0:  ff000000 3fff22f4 0019001f ff00746e  
3ffffeb0:  3fff2254 0039003f ff000000 00000000  
3ffffec0:  00000000 00000001 00000000 00000000  
3ffffed0:  00000000 00000031 00000000 00fffe61  
3ffffee0:  6470752f 00657461 00ff0678 54534f50  
3ffffef0:  00000000 00ffff10 3fff1e5c 0000006f  
3fffff00:  ff746164 00000007 000089e9 3fff089c  
3fffff10:  3fff0678 00000001 00000000 00000001  
3fffff20:  00000000 3fff0704 00000000 40202e0c  
3fffff30:  00000000 3fff19b4 3fff0650 3fff089c  
3fffff40:  00000001 3fff0678 3fff0650 40204713  
3fffff50:  4020b130 00000000 00001388 3fffefa0  
3fffff60:  00000000 3fff19b4 0000000a 40208e23  
3fffff70:  3fffdad0 00000000 3fff086c 40201143  
3fffff80:  00000000 00000000 ffefeffe feefeffe  
3fffff90:  00000000 00000000 00000001 3fff089c  
3fffffa0:  3fffdad0 00000000 3fff086c 402088fc  
3fffffb0:  feefeffe feefeffe 3ffe8514 40100535  
<<<stack<<<
```

luffykesh, 2019-03-11:
I converted the ESPExceptionDecoder to CLI version and successfully decoded the stack.
Here is the [gist](https://gist.github.com/luffykesh/616e869a69ae310963dd57869b8f624c) which contains stack trace along with the the decoding.
```cpp
Decode Success
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4000228b
EXCVADDR: 0x4024549b

Decoding stack results
0x40208dd0: ets_printf_P(char const*, ...) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ core_esp8266_postmortem.cpp  line 80
0x40208dc4: ets_printf_P(char const*, ...) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ core_esp8266_postmortem.cpp  line 80
0x40100290: vPortFree(void*, char const*, int) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ heap.cpp  line 61
0x40208e1e: print_stack(uint32_t, uint32_t) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ core_esp8266_postmortem.cpp  line 197
0x40208fc4: __wrap_system_restart_local() at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ core_esp8266_postmortem.cpp  line 175
0x4020aa9e: umm_realloc(void*, size_t) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/ umm_malloc.cpp  line 1745
0x4020973c: realloc_loc(void*, size_t, char const*, int) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ heap.cpp  line 142
0x4020aa9e: umm_realloc(void*, size_t) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/ umm_malloc.cpp  line 1745
0x4020973c: realloc_loc(void*, size_t, char const*, int) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ heap.cpp  line 142
0x40202cd4: WiFiUDP::stopAll() at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi/src/ WiFiUdp.cpp  line 270
0x40206113: std::_Function_handler<void(), ESP8266HTTPUpdateServer::setup(ESP8266WebServer*, const String&, const String&, const String&)::__lambda2>::_M_invoke(const std::_Any_data &) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266HTTPUpdateServer/src/ ESP8266HTTPUpdateServer.cpp  line 72
0x402079b0: String::~String() at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ WString.cpp  line 125
0x402044e2: std::function<void ()>::operator()() const at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/ functional  line 2465
0x40202d34: FunctionRequestHandler::canUpload(String) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/detail/ RequestHandlersImpl.h  line 30
0x4020455d: FunctionRequestHandler::upload(ESP8266WebServer&, String, HTTPUpload&) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/detail/ RequestHandlersImpl.h  line 51
0x40207ce0: String::String(String const&) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ WString.cpp  line 41
0x4020522b: ESP8266WebServer::_parseForm(WiFiClient&, String const&, unsigned int) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/ Parsing.cpp  line 479
0x40204b31: ESP8266WebServer::_parseArguments(String const&) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/ Parsing.cpp  line 299
0x40204cd0: std::_Function_handler<void (String&, String&, String const&, int, int, int, int), storeArgHandler>::_M_invoke(std::_Any_data const&, String&, String&, String const&, int, int, int, int) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/ functional  line 2069
0x40204530: FunctionRequestHandler::upload(ESP8266WebServer&, String, HTTPUpload&) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/detail/ RequestHandlersImpl.h  line 46
0x40202d34: FunctionRequestHandler::canUpload(String) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/detail/ RequestHandlersImpl.h  line 30
0x40205ba9: ESP8266WebServer::_parseRequest(WiFiClient&) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/ Parsing.cpp  line 210
0x40202e88: FunctionRequestHandler::canHandle(HTTPMethod, String) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/detail/ RequestHandlersImpl.h  line 20
0x4020478f: ESP8266WebServer::handleClient() at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WebServer/src/ ESP8266WebServer.cpp  line 303
0x4020911f: delay(unsigned long) at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ core_esp8266_wiring.cpp  line 54
0x40201143: loop() at /Users/rushikesh/Documents/Arduino/sketch_apr30a/ sketch_apr30a.ino  line 46
0x40208bf8: loop_wrapper() at /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/ core_esp8266_main.cpp  line 125
```

hreintke, 2019-03-11:
Sorry for using this thread to get other info.
@luffykesh : would also like to have a CLI version of ESPExceptionDecoder .
Do you have it somewhere available or can you show how it can be converted ?

luffykesh, 2019-03-11:
@hreintke https://github.com/luffykesh/EspExceptionDecoder

d-a-v, 2019-03-11:
@luffykesh 

Can you try commenting this line:
https://github.com/esp8266/Arduino/blob/192aaa42919dc65e5532ea4b60b002c4e19ce0ec/libraries/ESP8266WiFi/src/WiFiUdp.cpp#L269

luffykesh, 2019-03-11:
@d-a-v 
Yes, It works when 
https://github.com/esp8266/Arduino/blob/192aaa42919dc65e5532ea4b60b002c4e19ce0ec/libraries/ESP8266WiFi/src/WiFiUdp.cpp#L269
is commented.
The __func__ macro is what is causing the problem. I tried printing it elsewhere in the sketch too.

luffykesh, 2019-03-12:
@d-a-v I have 3 devices, 2 of which do not boot after OTA is successful (after removing debug print in WiFiUdp::stopAll), with CORE debug on.
Just simply stops here.
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v2528013c
~ld
e:
 ets Jan  8 2013,rst cause:3, boot mode:(3,6)

ets_main.c 
```

d-a-v, 2019-03-12:
It's a different issue and OP is about to be closed-fixed.
Since several firmware versions are now around, can you try and flash erase those two devices, reflash them then OTA-flash again and report ? If this new issue persists, please file a new bug.

PR #5728 (2019-03-14): fix DEBUG macros
All fmt strings in flash
fix #5658

This also allows to avoid warnings and easy mistakes with (no brace):
```
    if (something)
        DEBUGV("blah");
```

*edit*
use newlib unaligned-compatible printf for DEBUGV
fix #5855
closed 2019-03-14 by d-a-v