esp8266/Arduino #5856: ESP8266 slows down after a while, then crashes

Zunzulla, 2019-03-09:
Hardware:                                   ESP8266-12E/F
Core Version:                              Latest core version
Development Environment:       Arduino IDE
Operation System:                     Windows 10

Board:  			NodeMCU 1.0 (ESP-12E Module)
Upload Speed:        115200
CPU Frequency :     "80MHz
Flash Size:               "4M (3M SPIFFS)"

**The Project:** A row of ESP8266 modules, that pass on a message

**The Problem:** The chips manage to pass on a message for 15-20min., then one starts slowing down till he crashes.

**Code of one of the crashing Chips:**

`  #include <ESP8266WiFi.h>

  #define       LED0      2       

  char*         ssid;
  char*         password;
  IPAddress     IServer(192,168,7,1);

  WiFiClient    IClient;

  #define       MAXSC     2

  WiFiServer    IchServer(8000);
  WiFiClient    IchClient[MAXSC];

  IPAddress     EineIP(192,168,6,1);
  IPAddress     NMask(255,255,255,0);  
  
  void setup() 
  {
    Serial.begin(115200);     
    pinMode(13,OUTPUT);         
    pinMode(LED0, OUTPUT);          
    digitalWrite(LED0, !LOW);       

    Serial.println("I/O Pins Modes fertig...");

    WiFi.mode(WIFI_AP_STA);           
    //WiFi.disconnect();
    delay(10);
    
    WiFi.begin("HALLO213");        

    Serial.println("!--- Verbindungsaufbau mit " + WiFi.SSID() + " ---!");

    
    VerbindungTest();           

    digitalWrite(LED0, !HIGH);  
    Serial.println("!-- Verbunden --!");

    Serial.println("Verbunden mit     : " + String(WiFi.SSID()));
    Serial.println("Signalstärke  : " + String(WiFi.RSSI()) + " dBm");
    Serial.print  ("Server IP Adresse : ");
    Serial.println(IServer);
    Serial.print  ("Eigene IP Adresse : ");
    Serial.println(WiFi.localIP());
    delay(1000);
  
    TKDRequest();      
    
    SetWifi("HALLO321","");
  }

  
  void loop()
  {  
    IrgendEinName();
  }

  void SetWifi(char* Name, char* Password)
  { 
    Serial.println("WIFI Mode : AccessPoint Station");
    
    ssid      = Name;                
    password  = Password;
    WiFi.softAPConfig(EineIP, EineIP, NMask);
    WiFi.softAP(ssid, password);
    Serial.println("WIFI < " + String(ssid) + " > ... Started");
    
    delay(1000);
    IPAddress IP = WiFi.softAPIP();    
    Serial.print("AccessPoint IP : ");
    Serial.println(IP);

    IchServer.begin();                  
    Serial.println("Server Started");
    Serial.println(WiFi.softAPIP());
  }

  void VerbindungTest()
  {
    while(WiFi.status() != WL_CONNECTED)
    {
      for(int i=0; i < 10; i++)
      {
        digitalWrite(LED0, !HIGH);
        delay(250);
        digitalWrite(LED0, !LOW);
        delay(250);
        Serial.print(".");
      }
      Serial.println("");
    }
  }

  void TKDRequest()
  {
    //IClient.stop();        

    if(IClient.connect(IServer, 8300))   
    {
      IClient.flush();
      Serial.println    ("<Verbunden>");
      IClient.println ("<Verbunden>");
    }
  }

  void IrgendEinName()
  {
    if (IchServer.hasClient())           
    {
      Serial.println("gefunden");
      for(int i = 0; i < MAXSC; i++)  
      {                                                          
        if (!IchClient[i] || !IchClient[i].connected())              
        {                                                       
          if(IchClient[i]) IchClient[i].stop();                       
          IchClient[i] = IchServer.available();                     
          Serial.print("Neuer Client : "); Serial.print(String(i+1) + " - ");
          continue;                                                   
        }
      }
      digitalWrite(LED0, HIGH);                
      WiFiClient IchClient = IchServer.available();
      //IchClient.stop();
    }
      //Serial.println("draußen");    
    for(int i = 0; i < MAXSC; i++)                 
    {
      //Serial.println("Unten");
      if (IchClient[i] && IchClient[i].connected())     
      {
        if(IchClient[i].available())                   
        {
          while(IchClient[i].available())         
          {
            String Nachricht = IchClient[i].readStringUntil('\n');
            Serial.println(Nachricht); 

            IchClient[i].flush();
            IchClient[i].println("<OK>");                       
            IClient.flush();
            IClient.println(Nachricht);
            digitalWrite(13,HIGH);
            delay(500);
            digitalWrite(13,LOW); 
          }
        }
      }
    }
  }`

**Attempts:** I went through 25 chips, all of them crashed. I also tried changing the power supply without success. Looked through the internet for people with similar problems, but none of the solutions worked for me (Putting a capacitor across GND and 5V+, using a different protocol (UDP, ESP NOW etc.), cooling the chip)

TD-er, 2019-03-11:
What are you checking with this `if(IchClient[i])` ?
It if a `WiFiClient`, not a pointer to one.
So I guess your buffers are filling up with received packets but you're not reading them.

jandrassy, 2019-03-11:
> 
> 
> What are you checking with this `if(IchClient[i])` ?
> It if a `WiFiClient`, not a pointer to one.
> So I guess your buffers are filling up with received packets but you're not reading them.

WiFiClient::operator bool()
{
    return connected();
}

devyte, 2019-03-11:
Closing due to user error.

closed 2019-03-11 by devyte

Zunzulla, 2019-03-20:
Sorry I forgot that I've also posted this question here. I managed to figure it out myself by adding the yield() method at the end of the loop. This solved the problem

Zunzulla, 2019-03-20:
> What are you checking with this `if(IchClient[i])` ?
> It if a `WiFiClient`, not a pointer to one.
> So I guess your buffers are filling up with received packets but you're not reading them.

It checks whether data is available or not

jandrassy, 2019-03-21:
> > What are you checking with this `if(IchClient[i])` ?
> > It if a `WiFiClient`, not a pointer to one.
> > So I guess your buffers are filling up with received packets but you're not reading them.
> 
> It checks whether data is available or not

it does not. It checks if the client object is valid. Different libraries have different checks. ESP8266WiFiLibrary checks if it is connected().

d-a-v, 2019-03-21:
https://arduino-esp8266.readthedocs.io/en/2.5.0/esp8266wifi/client-examples.html#general-client-loop

@Zunzulla Where exactly did you put your `yield()` ?