esp8266/Arduino #5857: AsyncMQTT & OTA does not work together

rohitjust24idpl, 2019-03-09:
Hi @igrr @marvinroger @me-no-dev @d-a-v, 

### Basic Infos
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			        [ESP-12]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB/3MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
Using following libraries
MQTT: [AsyncMQTT](https://github.com/marvinroger/async-mqtt-client) **(OVER SSL)**
OTA: ESP8266httpUpdate

Observations as follows

[image]

**Scenario 1**
1. As per the comments in [#1676](https://github.com/esp8266/Arduino/issues/1676),I tried to tweak the settings and lwip v1.4 worked for me for OTA.
2. But MQTT crashes when try to connect and throws exception (29).

**Scenario 2**
1. MQTT works on lwip v2. OTA also works on select ISP where network is stable. 
2. But throws Stream Read Timeout while updating in few ISPs where network is unstable.

**NOTE:** I am sure this is not a memory issue. I am using the example sketches mentioned in the libraries.

Currently I am stuck how to get these two things work together. 
Looking forward for your response. Any help is appreciated. 

### Debug Messages
```
Exception (29):
epc1=0x40210242 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000
```

`Update error: ERROR[6]: Stream Read Timeout
`

dmdt, 2019-03-09:
Seems I have same problem [#5818](https://github.com/esp8266/Arduino/issues/5818#issue-413867792). But I trying update clear ESP over HTTPS and also getting read timeout. Over HTTP updates ok. Using lwip v1.4 lm.

devyte, 2019-03-12:
@rohitjust24idpl said:
> But throws Stream Read Timeout while updating in few ISPs where network is unstable.

If memory serves, there is a timeout somewhere for Streaming, which signifies max time allowed between two consecutive read attempts.
What happens if you increase that timeout?

rohitjust24idpl, 2019-03-13:
@devyte 

What happens if you increase that timeout?

**ESP8266httpUpdate.cpp**: Tried to increase this timeout does not works.
```
ESP8266HTTPUpdate::ESP8266HTTPUpdate(void)
        : _httpClientTimeout(8000)
{
}
```

**Stream.cpp**: Tried to increase this timeout does not works.
```
void Stream::setTimeout(unsigned long timeout)  // sets the maximum number of milliseconds to wait
{
    _timeout = timeout;
}
```

**Updater.cpp** : I see here it attempts twice to read bytes before assigning UPDATE_ERROR_STREAM error. I tried increasing the number of attempts or timeout in data.readbytes both. But it doesn't seems to work. Your thoughts ?

```
    while(remaining()) {
        toRead = data.readBytes(_buffer + _bufferLen,  (_bufferSize - _bufferLen));
        if(toRead == 0) { //Timeout
            delay(100);
            toRead = data.readBytes(_buffer + _bufferLen, (_bufferSize - _bufferLen));
            if(toRead == 0) { //Timeout
                _currentAddress = (_startAddress + _size);
                _setError(UPDATE_ERROR_STREAM);
                _reset();
                return written;
            }
        }
        _bufferLen += toRead;
        if((_bufferLen == remaining() || _bufferLen == _bufferSize) && !_writeBuffer())
            return written;
        written += toRead;
        yield();
    }
```

Please guide me which timeout to set ?

Swiftnesses, 2019-04-09:
I'm also interested in knowing this as I have issues with V2...

b-rohit, 2019-09-16:
I also have the same issue on ESP8266 nodemcu and Wimex.

earlephilhower, 2020-07-18:
Closing due to no MCVE, no decoded stack dump, and age.  There's unfortunately nothing we can do with the little info in this issue.

closed 2020-07-18 by earlephilhower