esp8266/Arduino #5866: I can't use OTA update when ESP is an Access Point

OladapoAjala, 2019-03-11:
### Basic Infos
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]

### Problem Description
I can't perform Arduino OTA update on an ESP serving as the access point on my network. The IDE fails to detect the network port when my laptop is connected to the network created by my ESP module. OTA works fine when I set the ESP as a station and connect it to another network with my laptop connected to that network also.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#ifndef STASSID
#define STASSID "my-ssid"
#define STAPSK  "my-password"
#endif

IPAddress apIP(192, 168, 1, 1);
const char* ssid = STASSID;
const char* password = STAPSK;

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP(ssid, password);

  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}

bigFin, 2019-03-12:
I had the same issue and was able to do OTA update in AP mode by commenting out the includes for ESP8266mDNS.h and WiFiUdp.h

bigFin, 2019-03-13:
+1 I have this issue on NodeMcu v1 (SILABS / ESP 12-E), Windows 10, Arduino IDE and VS Code, all latest on git

OladapoAjala, 2019-03-13:
What do you think is the solution to this @bigFin

bigFin, 2019-03-13:
I'm not sure, I was able to perform OTA on my previous build of the esp8266 core (not sure what that was), but not on the devices that I've flashed with the current git branch. I was temporarily able to perform an OTA update in hard AP mode with ESP8266mDns and WiFiUDP disabled but this isn't working now.

zacharydrew, 2019-03-19:
I just submitted a pull request that fixes the issue for me: [PR 5894](https://github.com/esp8266/Arduino/pull/5894)

Lan-Hekary, 2019-03-25:
More on this issue ..
I managed to reproduce it ..
I noticed that it was solved by setting the wifi mode to AP_STA ..

I was using the IP of the Device to upload ..
Not esp8266.local ..

bigFin, 2019-03-28:
@zacharydrew FYI, your pull request does not fix the issue for me.

zacharydrew, 2019-03-28:
@bigFin, agreed. It’s a partial fix at best. It works intermittently for me or sometimes not at all. Before this change it never worked however.

I did see somewhere you can still use the esptool tool upload the binary directly to the IP though.

bigFin, 2019-03-30:
@zacharydrew doesn't ESP8266FS utilize esptool.py?

zacharydrew, 2019-03-30:
Esptool to the numerical IP address rather than the mDNS esp8266.local address.

I hadn’t heard of ESP8266FS, I will have to take a look!

bigFin, 2019-03-31:
@zacharydrew neither the IP or the .local address work for me.

d-a-v, 2019-07-05:
There is an update for mDNS in current master.
Does it work now when AP IP address is used ?
```
mDNS.begin("esp8266", WiFi.softAPIP())
```
@Lan-Hekary

Lan-Hekary, 2019-07-05:
@d-a-v 
Sadly no ..

It does not work if `WiFi.mode(WIFI_AP);`
but it DOES word if `WiFi.mode(WIFI_AP_STA);`
even without `WiFi.begin` call ..

Here is the Code :
`
	#include <Arduino.h>
	#include <ESP8266WiFi.h>
	#include <ESP8266mDNS.h>
	#include <WiFiUdp.h>
	#include <ArduinoOTA.h>
	#include <ESP8266WebServer.h>

	ESP8266WebServer server(80);

	#ifndef STASSID
	#define STASSID "my-ssid"
	#define STAPSK "my-password"
	#endif

	IPAddress apIP(192, 168, 4, 1);
	const char* ssid = STASSID;
	const char* password = STAPSK;

	void setup() {
		Serial.begin(115200);
		Serial.println("Booting");

		WiFi.mode(WIFI_AP);
		WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
		WiFi.softAP(ssid, password);

		MDNS.begin("esp8266", WiFi.softAPIP());
		Serial.println("Ready");
		Serial.print("IP address: ");
		Serial.println(WiFi.softAPIP());

		ArduinoOTA.begin();
		server.on("/",[](){
			delay(100);
			Serial.println("Hello!");
			server.send(200,"text/plain", "Hello!!");
		});
		server.begin();

	}

	void loop() {
		ArduinoOTA.handle();
		server.handleClient();
		delay(50);
	}`

d-a-v, 2019-07-05:
Please try with `apIP` instead of `WiFi.softAPIP())`
I just tried and my android phone sees mDNS

d-a-v, 2019-07-05:
Well, it worked once. mDNS Debugging enabled shows strange IP address after some time.

Lan-Hekary, 2019-07-05:
can you tell me the name of the app you are using on your mobile phone ..

> Well, it worked once.

Yup .. I noticed that on my mobile phone I'm using an app Called "Service Browser"
once I reset the ESP8266 and Entry appears in the app .. but it does not appear again unless I reset the ESP ..

d-a-v, 2019-07-06:
Service Browser too. There is a bug in UDP, I'm currently investigating.

d-a-v, 2019-07-08:
It appears that
- after the UDP fix
- after applying [this IP address](https://github.com/esp8266/Arduino/pull/5529/files)

It works on my android phone every time (with "Service Browser" app).

Lan-Hekary, 2019-07-08:
I am waiting for the merge to test the UDP fix ..
but I am wandering about this specific IP !!
it does not make sense to me ..