esp8266/Arduino #5867: connection ap timed out;

kaihangchen, 2019-03-11:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8285 device|other] nodemcu v1]
- Core Version:			[2.5.0]
- Development Env:		Arduino IDE1.8
- Operating System:		[Windows10]

### Settings in IDE

- Module:  			[nodemcuv1]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   115200

### Problem Description

example esp8266wifimesh.ino 

```

### Debug Messages
connection ap timed out;
```
Debug messages go here
```
connection ap timed out;

devyte, 2019-03-11:
@aerlon does this make any sense to you?

aerlon, 2019-03-12:
@devyte Not much. `connection ap timed out` is not a message generated by the mesh library as far as I can tell, and I have no file named "esp8266wifimesh.ino". To make sure I just tried the `HelloMesh.ino` example with the latest master branch and everything works as expected.

@kaihangchen We are having trouble understanding your issue. Could you please provide a [MCVE](https://stackoverflow.com/help/mcve) code example that shows the problem? Also, if you have not already done so it may help fix your problem if you update your ESP8266 Arduino Core to the latest master branch version (see the "Using git version" heading in [the docs](https://arduino-esp8266.readthedocs.io/en/latest/installing.html)).

devyte, 2019-03-12:
@aerlon thanks for checking.
Closing due to lack of info. 
I'll reopen if a sane MCVE is provided that reproduces an issue in our core.

closed 2019-03-12 by devyte