esp8266/Arduino #5871: Error in mDNS code - schedule_function

git-tiger, 2019-03-13:
Error with build. 'schedule_function' was not declared in this scope.
Using platformio Espressif 8266 library (version 2.0.4).
 
> 
C:\Users\user\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src\LEAmDNS.cpp: In lambda function:
C:\Users\user\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266mDNS\src\LEAmDNS.cpp:100:76: error: 'schedule_function' was not declared in this scope
Asrchiving .pioenvs\RACETRACKER2\lib8a1\libServo.achedule_function(std::bind(&MDNSResponder::_restart, this));
>

devyte, 2019-03-13:
This hints at wrong installation. If it were true, CI would be failing the pio builds.
Closing.

closed 2019-03-13 by devyte

git-tiger, 2019-03-13:
It was building correctly days ago. Updated platformio, and it errorred.
There is a log change in the mDNS for version 2.5.0