esp8266/Arduino #5875: Signed binaries fail signature verification when doing OTA via Arduino IDE

imavroukakis, 2019-03-13:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-01
- Core Version:			SDK:2.2.1(cfd48f3)/Core:2.5.0-52-ge46ccae9=20500052/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-2-ga501b57/BearSSL:6778687
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			other
- Flash Size: 		1MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   other

### Problem Description

Uploading the same signed binary using OTA via the Arduino IDE fails with the following message
```
Progress: 100%
[Updater] sigLen: 1962934272
ERROR[12]: Signature verification failed
```

OTA updates with unsigned binaries are not affected, and the initial signed binary upload via OTA works fine.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecureBearSSL.h>
#include <ArduinoOTA.h>

const char* ssid     = "guest_intra";
const char* password = "xxxxxxxxxxx";
const uint8_t fingerprint[20] = {0xEE, 0xFE, 0x01, 0x7C, 0x5A, 0xCC, 0x76, 0x47, 0xEB, 0x30, 0x10, 0x84, 0xF8, 0xEA, 0x35, 0xB2, 0x53, 0x8F, 0x66, 0xAD};
const std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);

void setupOTA() {

  ArduinoOTA.setPasswordHash("th3h4sh");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
}
void setup() {
  Serial.begin(115200);
  delay(10);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.printf("\nConnecting to %s\n", ssid);

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  Serial.print('\n');

  Serial.print("Connection established!\n");
  Serial.printf("IP address:%s\n", WiFi.localIP().toString().c_str());
  setupOTA();
}

void loop() {
  ArduinoOTA.handle();
  if (Serial.available() > 0) {
    //do some other stuff
  }
}

```

### Debug Messages

```
Progress: 100%
[Updater] sigLen: 1962934272
ERROR[12]: Signature verification failed
Error[4]: End Failed
```

devyte, 2019-03-14:
@imavroukakis ESP-01 => does it have a Puya flash chip? If so, you need to build with Puya support enabled, otherwise onboard writing won't work correctly.

imavroukakis, 2019-03-15:
@devyte thanks for the tip. It is indeed a Puya and I've been reading up on the horror stories. Is what's described at https://github.com/esp8266/Arduino/issues/5493#issuecomment-452644552 what's required to enable support for it?

imavroukakis, 2019-03-15:
I added 
`platform.txt:compiler.c.extra_flags=-DPUYA_SUPPORT=1`
but it hasn't made a difference

imavroukakis, 2019-03-15:
So here's what I've done
in `platform.txt` I added
`build.extra_flags="-DPUYA_SUPPORT=1 -DESP8266"`

In the build output I can see
```
/Development/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__
 -DICACHE_FLASH -U__STRICT_ANSI__ -I/Development/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include
  -I/Development/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lwip2/include
   -I/Development/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include
    -I/var/folders/68/_vv68wgd47q_k4h_b6ryj63c0000gp/T/arduino_build_899653/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x
     c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0
      -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_OTA -DARDUINO=10900 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266
       "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -DPUYA_SUPPORT=1 -DESP8266
        -I/Development/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266
         -I/Development/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic
          /var/folders/68/_vv68wgd47q_k4h_b6ryj63c0000gp/T/arduino_build_899653/sketch/wifi.ino.cpp -o /dev/null
```

and for good measure, added `#define PUYA_SUPPORT=1` in my sketch, unfortunately I'm still getting signature verification failure

btw, I'm not using SPIFFS at all.

imavroukakis, 2019-04-15:
Hey :) any updates/suggestions about this ?

earlephilhower, 2019-07-24:
#6334 fixed this, as the problem was espota.py was taking the unsigned binary, not the signed one, by default.  @thangktran fixed it to check and upload the xxx.signed version, should it exist, automatically.

closed 2019-07-24 by earlephilhower