esp8266/Arduino #5877: Can't use HTTPClient for multiple methods in class

dmdt, 2019-03-13:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE/VSCode extension]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu v1.0]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I have class with several method, each of them should get data from server. I'm declaring
BearSSL::WiFiClientSecure pointer (as uniqe_ptr/regular pointer/object, commented rows in code) and getting Exception (9) or Exception(28) and wdt reset (decoded stack below). As I understand Exception (9) appears cause I declaring uniqe_ptr in constructor, and after exiting constructor unique_ptr deletes object. But when I use regular pointer - nothig changes ((9) -> (28)).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecureBearSSL.h>

class check {
    public:
    // BearSSL::WiFiClientSecure* client = new BearSSL::WiFiClientSecure;
    // BearSSL::WiFiClientSecure client1;
    std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure); 
    HTTPClient* http;
    check();
    void start();
};

check::check() {
    // std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure); 
    // BearSSL::WiFiClientSecure client1;
    // BearSSL::WiFiClientSecure* client = new BearSSL::WiFiClientSecure;
    client->setFingerprint("69:D6:81:6C:C9:B3:2F:78:C4:66:44:36:A5:79:49:99:DC:A9:DD:1F");
    // client1.setFingerprint("69:D6:81:6C:C9:B3:2F:78:C4:66:44:36:A5:79:49:99:DC:A9:DD:1F");
    HTTPClient https;
    String url = "https://host/request.php?a=1&b=2";
    https.begin(*client,url);
    // https.begin(client1,url);
    http = &https;
}

void check::start() {
    http->GET();
    String payload;
    payload = http->getString();
    Serial.println(payload);
}

void setup() {
    Serial.begin(115200);
    Serial.println();
    Serial.println("Connecting to wifi");
    WiFi.begin("Nodule","V1V@LARB");
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }
    Serial.println("Connected");
    Serial.println();
}

void loop() {
    check obj;
    obj.start();
    delay(5000);
}

```

### Decoded Stack

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40207ff4: IPAddress::fromString4(char const*) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\IPAddress.cpp line 93
EXCVADDR: 0x00000000

Decoding stack results
0x402029c3: ESP8266WiFiGenericClass::hostByName(char const*, IPAddress&, unsigned int) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 540
0x402041dc: std::__shared_ptr ::operator=(std::__shared_ptr &&) at c:\users\dmtprs\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/shared_ptr_base.h line 858
0x4010054c: _umm_free at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\umm_malloc\umm_malloc.c line 1300
0x40202a28: ESP8266WiFiGenericClass::hostByName(char const*, IPAddress&) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 530
0x40204c12: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 216
0x40207f70: HTTPClient::connected() at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 448
0x40205bac: HTTPClient::connect() at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 1055
0x40206388: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 591
0x40206436: HTTPClient::GET() at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 524
0x4020276c: check::start() at C:\Repository\Development\Arduino\Abug/Abug.ino line 23
0x4020283f: loop() at C:\Repository\Development\Arduino\Abug/Abug.ino line 44
0x40206b1c: Print::println() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\Print.cpp line 176
0x40207a00: loop_wrapper() at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 122

```

jandrassy, 2019-03-13:
http points nowhere after https goes out of scope

dmdt, 2019-03-13:
> http points nowhere after https goes out of scope

But when I declare https as class member? It shouldn't goes out of scope. 
And how I can fix it?

jandrassy, 2019-03-13:
make `HTTPClient https` a member. delete `HTTPClient* http`

devyte, 2019-03-13:
This is a "how do i" question. Closing.

closed 2019-03-13 by devyte